Ai-Thinker-Open/GPRS_C_SDK #322: Access phonebook from simcard using SDK

faithware, 2019-04-06:
## 1. SDK version（SDK 版本）

{

the latest one

}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{ I can't find a way to get to read sim card memory for phonebook contact.

}

---

Johhnzero, 2019-06-13:
SMS_GetStorageInfo，uset his。
#ifndef __API_SMD_H_
#define __API_SMD_H_

#include <sdk_init.h>

/// \brief  bool SMS_SetFormat(SMS_Format_t format, SIM_ID_t simID);
/// \param  SMS_Format_t format  , set the sms format
/// \param  SIM_ID_t simID  , set the sim id
/// \return bool , 
#define SMS_SetFormat                               CSDK_FUNC(SMS_SetFormat)

/// \brief  bool SMS_SetParameter(SMS_Parameter_t* smsParameter,SIM_ID_t simID);
/// \param  SMS_Parameter_t* smsParameter , get sms parameter
/// \param  SIM_ID_t simID  , set the sim id
/// \return bool , 
#define SMS_SetParameter                            CSDK_FUNC(SMS_SetParameter)

/// \brief  bool SMS_SendMessage(const char* phoneNumber, uint8_t* message,  uint8_t pduLength, SIM_ID_t simID);
/// \param  const char* phoneNumber  , the phone number sent sms to 
/// \param  uint8_t* message  , the meassage sented 
/// \param  uint8_t length  , length
/// \param  SIM_ID_t simID  , set the sim id
/// \return bool , 
#define SMS_SendMessage                             CSDK_FUNC(SMS_SendMessage)

#define SMS_SetServerCenterInfo                     CSDK_FUNC(SMS_SetServerCenterInfo)
#define SMS_GetServerCenterInfo                     CSDK_FUNC(SMS_GetServerCenterInfo)
#define SMS_ListMessageRequst                       CSDK_FUNC(SMS_ListMessageRequst)
#define SMS_DeleteMessage                           CSDK_FUNC(SMS_DeleteMessage)
#define SMS_GetStorageInfo                          CSDK_FUNC(SMS_GetStorageInfo)
#define SMS_SetNewMessageStorage                    CSDK_FUNC(SMS_SetNewMessageStorage)

/// \brief  const char* SMS_GetCharset(Charset_t);
/// \param  HANDLE hSem  , 
/// \return const char* , get the charset
#define SMS_GetCharset                              CSDK_FUNC(SMS_GetCharset)

#define SMS_Unicode2LocalLanguage                   CSDK_FUNC(SMS_Unicode2LocalLanguage)
#define SMS_LocalLanguage2Unicode                   CSDK_FUNC(SMS_LocalLanguage2Unicode)

#endif

Makimars, 2025-03-08:
Hi, I have the same issue, but the previous answer does not answer it. I want to read the phone numbers from the phonebook on the sim card, not the sms storage. It definitely is possible hardware-wise because the AT module can do it, but how do I do it on embedded?