Ai-Thinker-Open/GPRS_C_SDK #323: MQTT_SSL and AWS Broker (Problem)

AguilaStorm, 2019-04-07:
## 1. SDK version（SDK 版本）

{

SDK 2.112 
GPRS_C_SDK_V2112.7z
I have successfully compiled some examples: gpio, network, network_2

}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{

SO: Windows 10
Example problem: mqtt_ssl
Broker: AWS

The example of the version does not run the network, so I updated only the example to last commit 586b7c7, and the network apparently works because in the traces appears  "network activate success...".

Then, when run the MQTT conexion in the traces appears:

start mqtt test
MQTT connection status: 0
MQTT succeed connect to broker
MQTT OnMqttConnection() end
MQTT connected; now subscribe topic:app
MQTT connection status:256
connect to broker fail, error code: 256
MQTT OnMqttConnection() end
MQTT not connected to broker! can not publish

I suspected the connection with the broker. However, I checked the conexion with MQTT.fx V1.7.1 and there I completed the requested information (BrokerAddress, Broker Port, Client ID, SSL/TLS: CA File, Client Certificate File, Client Key File, Client Key Pasword, Protocol: TLSv1.2, PEM Formatted: True). The connection there was successful.

This is the information that I complete in the code:

    ci.client_id = "CLIENT ID";
    ci.client_user = CLIENT_USER;
    ci.client_pass = CLIENT_PASS;
    ci.keep_alive = 60;
    ci.clean_session = 1;
    ci.use_ssl = true;
    ci.ssl_verify_mode = MQTT_SSL_VERIFY_MODE_NONE;
    ci.ca_cert = ca_crt;
    ci.ca_crl = NULL;
    ci.client_cert = client_crt;
    ci.client_key  = client_key;
    ci.client_key_passwd = "KEY PASSWD";
    ci.broker_hostname = BROKER_HOSTNAME;
    ci.ssl_min_version   = MQTT_SSL_VERSION_SSLv3;
    ci.ssl_max_version   = MQTT_SSL_VERSION_TLSv1_2;
    ci.entropy_custom    = "GPRS_A9";
    ci.will_qos = 0;
    ci.will_topic = NULL;
    ci.will_retain = 0;
    memcpy(strstr(willMsg,"GPRS")+5,imei,15);
    ci.will_msg = willMsg;

Does anyone know the correct connection for a broker like AWS?

Thanks so much for any help!

}

---

AguilaStorm, 2019-04-17:
Hi!

I checked again it  and I found that the problem was the "MQTT_Publish" and "MQTT_Subscribe".

You should review the parameters for your application according to the [MQTT documentation](https://ai-thinker-open.github.io/GPRS_C_SDK_DOC/en/c-sdk/function-api/mqtt.html).

### MQTT_Publish
- client:MQTT client object
- topic: topic
- port: server port
- payload: message body
- payloadLen: message body length
- DUP: indicating the number of sending repeats
- QoS: quality of service
- retain: needs the server to keep the message persisted
- callback: publish request callback function
- Arg: parameters that need to be passed to the callback function
### MQTT_Subscribe
- client:MQTT client object
- topic: topic
- QoS: quality of service
- callback: subscribe to the topic callback function
- Arg: parameters that need to be passed to the callback function

I confirm that everything works perfectly.

closed 2019-04-17 by AguilaStorm

lvdiev, 2019-08-24:
Hello,
I'm facing the same problem with status 256.
Could you please share your working demo?

Thanks.

malte777, 2020-05-17:
Hi,
I also have this problem with status 256. With MQTT-Fx I can connect. But not with the A9G. I changed the QoS levels from 2 to 1. But it still doesn't work. What else has to be changed?

malte777, 2020-05-18:
I have it running now, by making these changes:
change
    ci.ssl_min_version   = MQTT_SSL_VERSION_SSLv3;
to
    ci.ssl_min_version   = MQTT_SSL_VERSION_TLSv1_2;

delete:
    ci.entropy_custom    = "GPRS_A9";
    ci.will_qos = 2;
    ci.will_topic = "will";
    ci.will_retain = 1;
    memcpy(strstr(willMsg,"GPRS")+5,imei,15);
    ci.will_msg = willMsg;