Ai-Thinker-Open/GPRS_C_SDK #324: FS bug

pulkin, 2019-04-08:
Using FS API:
1. Create /new_folder_1
2. Create /new_folder_1/new_folder_2
3. Create /new_folder_1/new_folder_2/new_file and write something into it
4. Remove /new_folder_1/new_folder_2/new_file
5. Removing /new_folder_1/new_folder_2 does not work any longer: it can be removed only after resetting the module

Johhnzero, 2019-06-13:
You mean the function to delete folders will only take effect after restart. Which function do you use?

pulkin, 2019-06-13:
> You mean the function to delete folders will only take effect after restart.

To be exact, this function has no effect before restart. It has to be called after restart to remove folder_2. The function is `API_FS_Mkdir(path, 0)` the way it is implemented in micropython:
```c
mp_obj_t mp_vfs_mkdir(mp_obj_t path_in) {
    const char* path = mp_obj_str_get_str(path_in);
    if(strcmp(path,"/")==0 || strcmp(path,"/t")==0)
    {
        mp_raise_OSError(MP_EEXIST);
    }
    int32_t ret = API_FS_Mkdir(path,0);
    if(ret != 0)
        mp_raise_OSError(MP_EIO);
    return mp_const_none;
```

Johhnzero, 2019-06-13:
You mean that the function that generates folders is only valid if it is called after reboot, rather than following your initial steps.

pulkin, 2019-06-13:
Sorry, the problem is ofc with the removal:

```c
mp_obj_t mp_vfs_rmdir(mp_obj_t path_in) {                                                                                                                                                                      
    const char* path = mp_obj_str_get_str(path_in);
    if(strcmp(path,"/")==0 || strcmp(path,"/t")==0)
    {
        mp_raise_OSError(MP_EINVAL);
    }
    char* path0 = (char*)malloc(160);
    memset(path0,0,160);
    API_FS_RealPath(path,path0);
    int32_t ret = API_FS_Rmdir(path0);
    free(path0);
    if(ret != 0)
        mp_raise_OSError(MP_EIO);
    return mp_const_none;
}
```