Ai-Thinker-Open/GPRS_C_SDK #325: /bin/bash: mips-elf-gcc: command not found

nhantt92, 2019-04-11:
## 1. SDK version（SDK 版本）

{

}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{

}

---

nhantt92, 2019-04-11:
[CC]   sdk_init.o                  <== sdk_init.c
/bin/bash: mips-elf-gcc: command not found
/home/nhantt/projects/GPRS_C_SDK/platform/compilation/cust_rules.mk:1005: recipe for target '/home/nhantt/projects/GPRS_C_SDK/build/gpio/init/obj/debug/sdk_init.o' failed
make[4]: *** [/home/nhantt/projects/GPRS_C_SDK/build/gpio/init/obj/debug/sdk_init.o] Error 127
/home/nhantt/projects/GPRS_C_SDK/platform/compilation/cust_rules.mk:834: recipe for target '/home/nhantt/projects/GPRS_C_SDK/build/gpio/init/lib/libinit_debug.a' failed
make[3]: *** [/home/nhantt/projects/GPRS_C_SDK/build/gpio/init/lib/libinit_debug.a] Error 2
/home/nhantt/projects/GPRS_C_SDK/platform/compilation/cust_rules.mk:1030: recipe for target 'dependencies' failed
make[2]: *** [dependencies] Error 2
/home/nhantt/projects/GPRS_C_SDK/platform/compilation/cust_rules.mk:748: recipe for target '/home/nhantt/projects/GPRS_C_SDK/build/gpio//gpio.srec' failed
make[1]: *** [/home/nhantt/projects/GPRS_C_SDK/build/gpio//gpio.srec] Error 2
/home/nhantt/projects/GPRS_C_SDK/platform/compilation/cust_rules.mk:395: recipe for target 'lod' failed
make: *** [lod] Error 2

pulkin, 2019-04-11:
Fix your .bashrc

nhantt92, 2019-04-12:
I edited the .bashrc file, then ./build.sh demo gpio

The following error appears: 

[MAKE] init

[CC] sdk_init.o <== sdk_init.c
as: unrecognized option '-G0'
/home/nhantt/projects/GPRS_C_SDK/platform/compilation/cust_rules.mk:1006: recipe for target '/home/nhantt/projects/GPRS_C_SDK/build/gpio/init/obj/debug/sdk_init.o' failed
make[4]: *** [/home/nhantt/projects/GPRS_C_SDK/build/gpio/init/obj/debug/sdk_init.o] Error 2
/home/nhantt/projects/GPRS_C_SDK/platform/compilation/cust_rules.mk:835: recipe for target '/home/nhantt/projects/GPRS_C_SDK/build/gpio/init/lib/libinit_debug.a' failed
make[3]: *** [/home/nhantt/projects/GPRS_C_SDK/build/gpio/init/lib/libinit_debug.a] Error 2
/home/nhantt/projects/GPRS_C_SDK/platform/compilation/cust_rules.mk:1031: recipe for target 'dependencies' failed
make[2]: *** [dependencies] Error 2
/home/nhantt/projects/GPRS_C_SDK/platform/compilation/cust_rules.mk:749: recipe for target '/home/nhantt/projects/GPRS_C_SDK/build/gpio//gpio.srec' failed
make[1]: *** [/home/nhantt/projects/GPRS_C_SDK/build/gpio//gpio.srec] Error 2
/home/nhantt/projects/GPRS_C_SDK/platform/compilation/cust_rules.mk:396: recipe for target 'lod' failed
make: *** [lod] Error 2
====== Build Time: 0s complete at 2019-04-11 13:30:24 =======
!!!!!!!!!!!!!!!!!!!!
BUILD FAILED
!!!!!!!!!!!!!!!!!!!!

pulkin, 2019-04-12:
This does not say anything to me. Your paths may still be broken, please take a look at my https://github.com/Ai-Thinker-Open/GPRS_C_SDK/issues/296#issuecomment-459745588.

arlucio, 2019-04-23:
I had the same problem while using a docker Ubuntu container.

To fix it, I just stopped the container and started again, it worked normally after.

Johhnzero, 2019-06-13:
What's your version?you can use v2.129

aabaenar, 2019-06-21:
@arlucio What's your SO, I have ubuntu 18.04 and have the problem whit [this instructions](https://ai-thinker-open.github.io/GPRS_C_SDK_DOC/en/c-sdk/installation_linux.html)