Ai-Thinker-Open/GPRS_C_SDK #326: as: unrecognized option '-G0'

nhantt92, 2019-04-11:
[MAKE]  init

[CC]   sdk_init.o                  <== sdk_init.c
as: unrecognized option '-G0'
/home/nhantt/projects/GPRS_C_SDK/platform/compilation/cust_rules.mk:1006: recipe for target '/home/nhantt/projects/GPRS_C_SDK/build/gpio/init/obj/debug/sdk_init.o' failed
make[4]: *** [/home/nhantt/projects/GPRS_C_SDK/build/gpio/init/obj/debug/sdk_init.o] Error 2
/home/nhantt/projects/GPRS_C_SDK/platform/compilation/cust_rules.mk:835: recipe for target '/home/nhantt/projects/GPRS_C_SDK/build/gpio/init/lib/libinit_debug.a' failed
make[3]: *** [/home/nhantt/projects/GPRS_C_SDK/build/gpio/init/lib/libinit_debug.a] Error 2
/home/nhantt/projects/GPRS_C_SDK/platform/compilation/cust_rules.mk:1031: recipe for target 'dependencies' failed
make[2]: *** [dependencies] Error 2
/home/nhantt/projects/GPRS_C_SDK/platform/compilation/cust_rules.mk:749: recipe for target '/home/nhantt/projects/GPRS_C_SDK/build/gpio//gpio.srec' failed
make[1]: *** [/home/nhantt/projects/GPRS_C_SDK/build/gpio//gpio.srec] Error 2
/home/nhantt/projects/GPRS_C_SDK/platform/compilation/cust_rules.mk:396: recipe for target 'lod' failed
make: *** [lod] Error 2
====== Build Time: 0s complete at 2019-04-11 13:30:24 =======
!!!!!!!!!!!!!!!!!!!!
   BUILD FAILED
!!!!!!!!!!!!!!!!!!!!
=============================================================

I have an error like this, please help everyone. thanks everyone

Johhnzero, 2019-06-13:
Compilation environment is not well configured, usually missing files

LubomirBogdanov, 2019-09-11:
> Compilation environment is not well configured, usually missing files

Hello, I have the same issue - but I believe that my environment is set up properly - if I type anywhere in a terminal (Linux Ubuntu 14.04 64-bit) mips-elf-gcc I see:

mips-elf-gcc: no input files

and if I type: 

echo $GPRS_PROJ_ROOT

I see:

/home/user/ai_thinker_projects

so the environment variable is properly set.

Any idea what might be causing this issue?

LubomirBogdanov, 2019-09-17:
OK, by using this link:
https://github.com/Ai-Thinker-Open/GPRS_C_SDK/issues/296

I typed in a terminal:
locate libcloog.so

and I got:

/home/user/git_projects/CSDTK/lib/libcloog.so.0
/home/user/git_projects/CSDTK/lib/libcloog.so.0.0.0
/home/user/programs/ai_thinker/CSDTK/lib/libcloog.so.0
/home/user/programs/ai_thinker/CSDTK/lib/libcloog.so.0.0.0

which means I have installed the CSDTK two times. I removed one of the copies and the build
succeeds now. 

I hope this helps!