Ai-Thinker-Open/GPRS_C_SDK #327: How to extract time and date from GPS data

andyman2240, 2019-04-11:
## 1. SDK version（SDK 版本）

{

}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{
I am trying to extract the time and date from the RMC sentence that is being received by the GPS receiver.    I am using a slightly modified version of the GPS demo  provided in GPRS_SDK_C.  The data is being properly recorded on the TF card as follows once GPS lock has been achieved, as shown in the following line:

$GNRMC,091835.000,A,2804.6123,N,08035.0009,W,1.828,204.06,110419,,,A*54

However when I try to display the time on an OLED display, I am getting RMCTime  displayed as 9 instead of the expected value of 091835.000 . I am using the following code to try and do this;

 snprintf(buffer4,sizeof(buffer4),"RMCTime: %c",gpsInfo->rmc.time);
OLED_ShowStr(5,7,buffer4,1);

I am by no means an experienced programmer, so I am asking for some help with this request. Thank you.

}

---

Lord-Lucan, 2019-04-15:
Not sure what your display device is but that snprintf format is not correct.
Looking at the rmc structure we have ....

struct minmea_time {
    int hours;
    int minutes;
    int seconds;
    int microseconds;
};

struct minmea_sentence_rmc {
    struct minmea_time time;

So the time is a structure, so you're either going to have to write some code to convert those hours,minutes,seconds,microsecnods to the string you want or at a pinch, this might work ...

snprintf(buffer4,sizeof(buffer4),"RMCTime: %d:%d:%d:%d",gpsInfo->rmc.time.hours,gpsInfo->rmc.time.minutes,gpsInfo->rmc.time.seconds,gpsInfo->rmc.time.microseconds);

At the moment the format character you have %c is a single character, probably not what you want. %d is the format character for an integer. So the time fields should be interpreted as integers.

Johhnzero, 2019-06-13:
%s