Ai-Thinker-Open/GPRS_C_SDK #329: Disconnect from GSM network, turn off modem

zkemble, 2019-04-21:
## 1. SDK version（SDK 版本）

{

V2.112

}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{

Is there anyway of disconnecting/unregistering from the GSM network and stop searching? And turn off the modem? `PM_SleepMode(true)` does not disconnect from the network, as mentioned here #79

}

---

punee995, 2019-04-22:
Here is the API for the Network.
You can simply search in Include folder for more API functionality.
https://github.com/Ai-Thinker-Open/GPRS_C_SDK/blob/master/include/api_network.h

zkemble, 2019-04-22:
Ahh thank you! I had been using the last release version which doesn't have a bunch of those functions.

closed 2019-04-22 by zkemble

medida, 2019-08-28:
@zkemble did you get the GSM to turn OFF and then back ON?

I cannot get it to turn back ON?

zkemble, 2019-08-28:
@medida This works for me:
`Network_Register((uint8_t[]){0,0,0,0,0,0}, NETWORK_REGISTER_MODE_MANUAL);`

However, there is a bug (again) where if you connect to GPRS, then deregister from the network and register again, GPRS will no longer be able to connect and might crash the module. Only restarting the module fixes it `PM_Restart();`.

medida, 2019-08-28:
thanks - i'll give that ago.

I really want to be able to put the device to sleep - turn off the modem / gps to save as much power as possible... but if i have to reboot i would have to save any settings and re-read them... therfore taking more time.

zkemble, 2019-08-28:
The lowest current consumption I can get the module down to is 1.6mA @ 4V (the datasheet says 1.46mA, so not too far off). To get any lower you will have to shutdown the module, which for me gives ~240uA (datasheet says 216uA), which is still kinda high, and you'll loose settings and things, and it can't automatically turn itself back on.
If want super low power down current you will need a microcontroller that can completely remove power from the module by controlling a P-MOSFET and communicate over I2C or something to store and retrieve settings.
An external microcontroller is a good idea anyway, as the SDK is super buggy and likely to crash or get stuck, so you'll need the controller to make sure the module is still ok and to reset it when it dies.

zkemble, 2019-08-28:
I just found that setting GPIO_PIN31, GPIO_PIN32, GPIO_PIN33 and GPIO_PIN34 to OUTPUT LOW reduces current by around 0.4mA, so now the lowest current draw while the module is on is 1.2mA.

medida, 2019-11-22:
I have found i can reconnect the network by calling the following code.

For some reason - if i try to connect with the wrong APN - then the correct one - it works!

  Network_PDP_Context_t context = {
                .apn        ="somethingwrongtomakeitfail",
                .userName   = "",
                .userPasswd = ""
            };
            Network_StartActive(context);
          
            Network_PDP_Context_t context2 = {
                .apn        ="mobiledata",
                .userName   = "",
                .userPasswd = ""
            };
         
            Network_StartActive(context2);
           //   Network_SetFlightMode(false); // worked when here
            OS_Sleep(3000);

zkemble, 2019-12-02:
Thanks for this @medida! I gave it a quick go and it seems to work!

General flow of things is:
1. Connect to GPRS with correct APN settings (Network_StartActive(context))
2. Wait for connection (API_EVENT_ID_NETWORK_ACTIVATED)
3. Disconnect from GPRS (Network_Deactive(1)) (I'm not sure what the context ID is supposed to be, I just put it in a 0 - 10 loop and ID 1 is usually the one that works)
4. Wait for disconnect (API_EVENT_ID_NETWORK_DEACTIVED)
5. Connect to GPRS with incorrect APN settings (Network_StartActive(contextBad))
6. Wait for connection failure (API_EVENT_ID_NETWORK_ACTIVATE_FAILED)
7. Connect to GPRS with correct APN settings (Network_StartActive(context))
8. Wait for connection (API_EVENT_ID_NETWORK_ACTIVATED)
9. Done!

ens4dz, 2021-09-13:
> I just found that setting GPIO_PIN31, GPIO_PIN32, GPIO_PIN33 and GPIO_PIN34 to OUTPUT LOW reduces current by around 0.4mA, so now the lowest current draw while the module is on is 1.2mA.

thanks for this info,
the lowest current after:

- removing gps antenna and close gps
- go to flightmode
- set cpu to 32khz 
- close uart1 

is 4mA
what did you do to get that 1.2mA !

medida, 2021-09-14:
I have no idea an more - i abandoned the product sometime ago as it was not working as i wanted and the project it was inteneded for stopped as well.