Ai-Thinker-Open/GPRS_C_SDK #333: Power off the modules when not needed.

harshitsihare, 2019-04-25:
hello i would like to use your product in low power application using gps and gprs, i need to use power management modes to save power, Can we switch off some modules like CAM and I2C when not needed? Can we control clocks to these modules. Can you please expose some API's for these purposes.

win32cpp, 2019-04-27:
Nobody, even the manufacturer, don't supports this product.

punee995, 2019-04-28:
@win32cpp manufacturer will not write a code for you.
If you know how to use documentation and how to code you'll definitely like the a9 and a9g.
@harshitsihare before creating new issue look at closed first #240

win32cpp, 2019-04-28:
@punee995 We do not need the code.
When there is a bug in the product, does anyone support it?
I talked hard to the manufacturer to fix a bug, which did not take any action to solve the problem at the end.

harshitsihare, 2019-04-29:
<<@harshitsihare before creating new issue look at closed first #240
@punee995 I am asking for powering off some modules selectively at a time not all together.
Secondly, Even after 'PM_SleepMode', it sets the frequency to 32kHz. But all GPIO pins remains active.
Similarly, If I use PM_PowerEnable with selective PAD, it does not power off that voltage line.

punee995, 2019-04-30:
Try this.

https://github.com/Ai-Thinker-Open/GPRS_C_SDK/blob/91832c812f2f285b46f723ef8dfb1b61a8e8661d/include/sdk_init.h#L48

harshitsihare, 2019-05-01:
@punee995 I have used     bool                (*PM_PowerEnable)(Power_Type_t powerType, bool isOn);
only.
It works well for powering on a line, but it does not power off any voltage rail. Did you try this already?, please share the example of using this to power off a voltage line. Thanks in advance!!

zkemble, 2019-05-01:
There are functions like I2C_Close() and SPI_Close() which turn off I2C/SPI modules, but whether or not they are internally powered down or if the clock is removed from them I don't know. You'll have to measure the current draw and see what happens with you have I2C_Init() and I2C_Close() in a slow loop.

Ricx8, 2019-05-02:
@harshitsihare I actually want the same

I guess you can't do anything about this GPIO
> GPIO0  ~ GPIO7  and GPIO25 ~ GPIO36 are allways on

This should work but I haven't tested. I'll tested this week. Please let me know if that works for you.
```
PM_PowerEnable(POWER_TYPE_MMC, false); // Power off from GPIO8  ~ GPIO13
PM_PowerEnable(POWER_TYPE_LCD, false); // Power off from GPIO14 ~ GPIO18
PM_PowerEnable(POWER_TYPE_CAM, false); // Power off from GPIO19 ~ GPIO24
```

Johhnzero, 2019-06-13:
There are some gpios that can't be turned off。GPIO0  ~ GPIO7  and GPIO25 ~ GPIO36    2.8V   //always on

typedef enum{
    POWER_TYPE_VPAD = 0,            // GPIO0  ~ GPIO7  and GPIO25 ~ GPIO36    2.8V   //always on
    POWER_TYPE_MMC,                 // GPIO8  ~ GPIO13                        1.9V
    POWER_TYPE_LCD,                 // GPIO14 ~ GPIO18                        1.9V
    POWER_TYPE_CAM,                 // GPIO19 ~ GPIO24                        1.9V
    POWER_TYPE_MAX
}Power_Type_t;

mnorval, 2019-08-16:
I have tested the **PM_Sleep** function where it sleeps for 1 hour (60 * 1 minute OS_Sleeps) and then wakes up.

_uint8_t tempcount=0;
PM_SleepMode(true);
Trace(1,"Enter deep sleep");
for (tempcount=0; tempcount < 60;tempcount++)
{
Trace(1,"Sleep %d",tempcount);OS_Sleep(60 * 1000);//1 minute
}
PM_SleepMode(false);
Trace(1,"Exit deep sleep");_

Battery consumption is still very high.
I suspect GSM remains active.

Is there a command to disable GSM?

zkemble, 2019-08-16:
> Battery consumption is still very high.
> I suspect GSM remains active.
> 
> Is there a command to disable GSM?

How high is very high?

GSM can be disabled using Network_DeRegister();. You will need the latest SDK stuff from the repo for it to work (v2.129).

mnorval, 2019-08-17:
Thank you Zak, I will give that a try.

zkemble, 2019-08-17:
You need to download the latest repo master (top right green button that says "Clone or Download").

mnorval, 2019-08-18:
I clone the folder with GIT, but every time I do so it is incomplete. Would it be possible to share completed files & folder structure with me on dropbox/sharing util?

zkemble, 2019-08-24:
Eh yea, looks like git clone isn't working properly (I'm getting some error with the libs/aliyun/iotkit-embedded submodule thing). I've forked the repo and made a new v2.129 release here https://github.com/zkemble/GPRS_C_SDK/releases/tag/v2.129

mnorval, 2019-08-27:
Thank you very much!

medida, 2019-08-28:
Don;t suppose this would work with the Network_SetFlightMode

Once the GSM has been turned off - how can you restart it without knowing the operator ID?

Can it just be called to restart automatically?

Thanks - its a shame there is no support from the manufacture.

medida, 2019-08-28:
@mnorval  - did you get the device to re-register to the network once you had deregistered it???

mnorval, 2019-08-28:
Thanks for the help medida. Anyone know how to obtain/extract the mobile number? EMEI is available.

mnorval, 2019-08-29:
What I do is just restart the module after sleeping for an hour or so.

mnorval, 2019-09-16:
Does anyone know how to obtain/extract the mobile number? Seems like only EMEI is available.

medida, 2019-11-22:
So i am currently getting info from AI-Thinker.

I have found a way to restart the GSM from flightmode without having to reboot the device.

I found that if i call the Network_startActive with an incorrect APN - then with the correct one it starts and connected correctly!!!!!!

  Network_PDP_Context_t context = {
                .apn        ="somethingwrongtomakeitfail",
                .userName   = "",
                .userPasswd = ""
            };
            Network_StartActive(context);
          
            Network_PDP_Context_t context2 = {
                .apn        ="mobiledata",
                .userName   = "",
                .userPasswd = ""
            };
         
            Network_StartActive(context2);
             Network_SetFlightMode(false); // worked when here
            OS_Sleep(3000);

luconedj, 2020-12-01:
> So i am currently getting info from AI-Thinker.
>
> I have found a way to restart the GSM from flightmode without having to reboot the device.
>
> I found that if i call the Network_startActive with an [...]

hello @medida ,
trying to accomplish the same but seems your WAS doesn't work for me.
Could you please share the way you call the Network_StartActive twice? is it in the Event switch case of the main task?
BR,
Luca