Ai-Thinker-Open/GPRS_C_SDK #335: git clone最新GPRS_C_SDK出错，子模块无法下载

ahut0316, 2019-05-04:
执行
git clone https://github.com/Ai-Thinker-Open/GPRS_C_SDK.git --recursive
无法下载子模块代码，报错信息：
error: no such remote ref e6bd3556039538ac504c005f0c31dbdd0e906658
Fetched in submodule path 'libs/aliyun/iotkit-embedded', but it did not contain e6bd3556039538ac504c005f0c31dbdd0e906658. Direct fetching of that commit failed.

niwantha33, 2019-05-05:
csdk folder always empty, downloaded several times but the same result was found in all the attempts.

Johhnzero, 2019-06-12:
直接链接下载，缺失的就单独下

swornim00, 2019-06-27:
I Separately downloaded the folder from [This](https://github.com/Ai-Thinker-Open/GPRS-C-SDK-LIB/tree/5040e04c3361de24fc8680b234b14e03cd9b0598) location and placed the files inside the csdk folder.

ariston1983, 2019-07-09:
There's several files/folders missing even use clone --recursive
- \libs\gps\minmea\src
- \platform\csdk
it need clone manually from submodule (not sure if this github issue or not)