Ai-Thinker-Open/GPRS_C_SDK #338: 请教recipe for target 'bin' failed

7730czh, 2019-05-11:
## 1. SDK version（SDK 版本）

{
V2112
}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{
CSDTK下载的最新版本CSDTK42_Update2_18_7_27.7z。
SDK文件下载最新版本压缩包文件GPRS_C_SDK_V2112.7z，通过Powershell窗口打开，通过./build.bat app编译文件，出现如下错误：
D:/GPRS_C_SDK/platform/compilation/cust_rules.mk:543: recipe for target 'bin' failed
make[1]: *** [bin] Error 181
D:/GPRS_C_SDK/platform/compilation/cust_rules.mk:371: recipe for target 'lod' failed
make: *** [lod] Error 2
请教一下，什么原因呢，非常感谢！

}

---

7730czh, 2019-05-11:
已解决。
经测试，win7系统重启后，可以编译成功。
win10系统在CMD模式下编译出现系统组件(msvcr100.dll)缺失的情况,修复后解决问题。

谢谢！

closed 2019-05-11 by 7730czh

kicsigabszi, 2022-12-22:
Hy @7730czh, 
the same issue happened by me. Can you explain please how you fixed this issue?

kicsigabszi, 2022-12-22:
this solution worked for me
https://github.com/Ai-Thinker-Open/GPRS_C_SDK/issues/180#issuecomment-409844287