Ai-Thinker-Open/GPRS_C_SDK #340: locating a stored sms

ichanjz, 2019-05-14:
## 1. SDK version（SDK 版本）

{
V2112

}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{
After reading a new incoming sms message under API_EVENT_ID_SMS_RECEIVED, I want to be able to choose whether to save or not to save it. So far, if I don't want the message, I could only delete it using SMS_DeleteMessage() after it's been stored. For that I need to SMS_ListMessageRequst() to read thru all the messages in store in order to find the unwanted message and obtain its 'index' which is required by SMS_DeleteMessage(). Does anyone know of a more elegant way of doing that and would like to share his/her knowledge here?

}

---

ichanjz, 2019-05-15:
I noticed later that someone had already asked for something similar in #319 which, apparently, did not attract any response so far. So I might as well close this one.

Johhnzero, 2019-06-13:
You mean you can't delete this message in API_EVENT_ID_SMS_RECEIVED？A little more detail to describe

ichanjz, 2019-06-13:
Sorry, not that I can't delete the message but only that I can't do that easily. Why? Simply because none of the event parameters contains the message id (or storage index). To delete it, my not so elegant method is to compare the timestamp of the new message to each one in store (50 in flash and 40 in sim) using SMS_ListMessageRequst() until a match (and hopfully only one match) is found, note its storage index, then delete it by SMS_DeleteMessage(). This, as I thought, is ridiculous while I should have been able to called SMS_DeleteMessage() when still handling the *_SMS_RECEIVED event only if the storage index were included in the event parameters. Someone already suggested that in #319 which unfortunately didn't arouse any interest.

Johhnzero, 2019-06-13:
https://github.com/Ai-Thinker-Open/GPRS_C_SDK/issues/342#issue-453160511。
You can refer to this for a try.

ichanjz, 2019-06-14:
Didn't try (the 'easy code' in #342) but haven't the least doubt that it should work, only to bear in mind that messageInfo.phoneNumber which your 'easy code' used to identify the target message is not a unique identifier of the messages in store and it might inadverdently delete some other wanted messages sent from the same phone number. However, given the latest v2112 of the SDK, if this is already the easiest and correct way to delete an sms message associated with an API_EVENT_ID_SMS_RECEIVED event, I definitely wouldn't call it a well-designed API. My reasons:
1. SMS_ListMessageRequst(), as I understood, consumes a lot of system resources. Each call to this api function triggers as many API_EVENT_ID_SMS_LIST_MESSAGE events as there are messages in store (40 in sim and 50 in flash, the documetned maximums). Having to call this api in order to obtain an identifier to delete a message is, in my opinion, definitely an abuse of system resources.
2. Once a message is stored, there seems to be no reliable (and yet easily comparable) identifier other than its store location index which, unfortunately, is not available when it triggers API_EVENT_ID_SMS_RECEIVED.