Ai-Thinker-Open/GPRS_C_SDK #342: delete sms fail

cryarchy, 2019-06-06:
Simple code:
```c
case  API_EVENT_ID_SMS_RECEIVED:
			Trace(1, "SMS Task: Message received!");
			if(pEvent->param1 == SMS_ENCODE_TYPE_ASCII) {
			 	Trace(1, "SMS: Message header: %s", pEvent->pParam1);
	            Trace(1, "SMS: Message content length: %d", pEvent->param2);
                Trace(1, "SMS: Message content: %s", pEvent->pParam2);
            } else {
            	Trace(1, "SMS: Message encoding is not ASCII!");
            }
			SMS_ListMessageRequst(SMS_STATUS_ALL, SMS_STORAGE_SIM_CARD);
			break;
		case API_EVENT_ID_SMS_LIST_MESSAGE:;
			SMS_Message_Info_t* messageInfo = (SMS_Message_Info_t*)pEvent->pParam1;
            Trace(1,"message header index:%d,status:%d,number type:%d,number:%s,time:\"%u/%02u/%02u,%02u:%02u:%02u+%02d\"", messageInfo->index, messageInfo->status,
                                                                                        messageInfo->phoneNumberType, messageInfo->phoneNumber,
                                                                                        messageInfo->time.year, messageInfo->time.month, messageInfo->time.day,
                                                                                        messageInfo->time.hour, messageInfo->time.minute, messageInfo->time.second,
                                                                                        messageInfo->time.timeZone);
            Trace(1,"message content len:%d,data:%s",messageInfo->dataLen,messageInfo->data);
            if (strcmp(messageInfo->phoneNumber, ADMIN_PHONE_NUMBER) == 0) {
				Trace(1, "Message from admin!");
			}
			if(!SMS_DeleteMessage(messageInfo->index, SMS_STATUS_ALL, SMS_STORAGE_SIM_CARD))
	            Trace(1,"delete sms fail");
	        else
	            Trace(1,"delete sms success");
            //need to free data here
            OS_Free(messageInfo->data);
            break;
```
[image]

Johhnzero, 2019-06-12:
Make sure that your UTI is correct, you can test the SMS_DeleteMessage function with official demo first, or send your code to me to test the cause of the error.

zkemble, 2019-08-09:
I think I've found that attempting to delete any SMS from the SIM too soon after receiving an SMS will fail. Waiting a few seconds and then deleting seems to be ok. Maybe the SIM is still busy doing stuff?
I've also found that attempting to delete an SMS from flash will delete the SMS, but then it ends up getting stuck in SMS_DeleteMessage().

Lord-Lucan, 2020-04-13:
I'm also having same issue. SMS_DeleteMessage just blocks indefinitely. Only way to get A9G back is to reset the pudding board. Any news on when this will be fixed?

locorcor, 2021-02-05:
hello everyone, is there any news regarding sms deleting?

jfbvm, 2021-09-07:
Hi guys,

Let me try to help a little.

With the firmware **Ai-Thinker_A9_A9G_AT_V02.02.20190915R.afw**
I wrote a function with the delete SMS commands and created a task to call this function every 3 seconds with a 2 seconds delay

```C
#include "string.h"
#include "stdio.h"
#include "buffer.h"
#include "gps_parse.h"
#include "math.h"
#include "gps.h"
#include "time.h"
#include "assert.h"

#include "api_os.h"
#include "api_debug.h"
#include "api_event.h"
#include "api_sms.h"
#include "api_hal_uart.h"
#include "api_gps.h"
#include "api_hal_pm.h"
#include "api_info.h"
#include "api_socket.h"
#include "api_network.h"
#include "api_hal_gpio.h"
#include "api_fs.h"
#include "api_lbs.h"

#define MAIN_TASK_STACK_SIZE                   (2048 * 2)
#define MAIN_TASK_PRIORITY                      0

/**
 * FUNCTIONS
 */
/**
 * @brief Clear SMS messages saved inside the chip
 * 
 */
void clear_sms(){
    OS_Sleep(3000);
    SMS_Storage_Info_t storageInfo;
    SMS_GetStorageInfo(&storageInfo,SMS_STORAGE_SIM_CARD);
    Trace(1,"[clear_sms]: sms storage sim card info, used:%d,total:%d",storageInfo.used,storageInfo.total);
    SMS_GetStorageInfo(&storageInfo,SMS_STORAGE_FLASH);
    Trace(1,"[clear_sms]: sms storage flash info, used:%d,total:%d",storageInfo.used,storageInfo.total);
    if(!SMS_DeleteMessage(5,SMS_STATUS_ALL,SMS_STORAGE_SIM_CARD))
        Trace(1,"[clear_sms]: delete sms fail");
    else
        Trace(1,"[clear_sms]: delete sms success");
}

/**
 * TASKS
 */
void EventDispatch(API_Event_t* pEvent)
{
    static uint8_t lbsCount = 0;
    switch(pEvent->id)
    {
        case API_EVENT_ID_NO_SIMCARD:
            Trace(1,"[EventDispatch]: NO SIM CARD%d!!!!",pEvent->param1);
            networkFlag = false;
            break;
        case API_EVENT_ID_SYSTEM_READY:
            Trace(1,"[EventDispatch]: system initialize complete");
            SYSTEM_INIT = true;
            flag |= 1;
[... 143 lines cut ...]
        if(OS_WaitEvent(gpsTaskHandle, (void**)&event, OS_TIME_OUT_WAIT_FOREVER))
        {
            EventDispatch(event);
            OS_Free(event->pParam1);
            OS_Free(event->pParam2);
            OS_Free(event);
        }
    }
}

void code_Main(void)
{
    gpsTaskHandle = OS_CreateTask(
        gps_MainTask,
        NULL, 
        NULL, 
        MAIN_TASK_STACK_SIZE, 
        MAIN_TASK_PRIORITY, 
        0, 
        0, 
        "gps tracker"
    );
    OS_SetUserMainHandle(&gpsTaskHandle);
}
```

mathematikoi, 2021-09-11:
hey @jfbvm doesn't it block indefinitely, in my case I wrote a for loop to iterate for 1 to 50 and delete all SMS in that range residing in flash memory, it deleted the first SMS and hanged indefinitely, did you have any luck with this?

jfbvm, 2021-09-11:
HI @mathematikoi,
In fact, I discovered that if you keep deleting the messages every time (any loop function), at some moment the board will block indefinitely. So the solution is to not create any loop function to delete messages, instead only clear the messages when a message arrives. But only this will may not help, you need to do like in SMS demo, you will need to create a listener API_EVENT_ID_SMS_LIST_MESSAGE to list all messages at the start of the program, on SMSInit set SMS_STORAGE_SIM_CARD for the storage mode (don't choose SMS_STORAGE_FLASH) and call the function ServerCenterTest, all like in SMS demo. Now the board seems very stable for SMSs. Please see the code below. Let me know if I forgot to write some variable or something else...

And let me ask you a question about another problem, did you already tested this board outside your house? I mean, in the street. Because I'm in the phase to test its connection outside my house, in the streets, and I based my code on the original GPS tracker demo code and it's very unstable for the "reconnecting" part. Are you facing any issues like these?

```C
#include "string.h"
#include "stdio.h"
#include "buffer.h"
#include "gps_parse.h"
#include "math.h"
#include "gps.h"
#include "time.h"
#include "assert.h"

#include "api_os.h"
#include "api_debug.h"
#include "api_event.h"
#include "api_sms.h"
#include "api_hal_uart.h"
#include "api_gps.h"
#include "api_hal_pm.h"
#include "api_info.h"
#include "api_socket.h"
#include "api_network.h"
#include "api_hal_gpio.h"
#include "api_fs.h"
#include "api_lbs.h"

#define MAIN_TASK_STACK_SIZE                   (2048 * 2)
#define MAIN_TASK_PRIORITY                      0

/**
 * FUNCTIONS
 */
void ServerCenterTest(){
    uint8_t addr[32];
    uint8_t temp;
    SMS_Server_Center_Info_t sca;
    sca.addr = addr;
    SMS_GetServerCenterInfo(&sca);
    Trace(1,"[ServerCenterTest]: server center address:%s,type:%d",sca.addr,sca.addrType);
    temp = sca.addr[strlen(sca.addr)-1];
    sca.addr[strlen(sca.addr)-1] = '0';
    if(!SMS_SetServerCenterInfo(&sca))
        Trace(1,"[ServerCenterTest]: SMS_SetServerCenterInfo fail");
    else
        Trace(1,"[ServerCenterTest]: SMS_SetServerCenterInfo success");
    SMS_GetServerCenterInfo(&sca);
    Trace(1,"[ServerCenterTest]: server center address:%s,type:%d",sca.addr,sca.addrType);
    sca.addr[strlen(sca.addr)-1] = temp;
    if(!SMS_SetServerCenterInfo(&sca)) Trace(1,"[ServerCenterTest]: SMS_SetServerCenterInfo fail");
    else Trace(1,"[ServerCenterTest]: SMS_SetServerCenterInfo success");
}
void SMSInit(){
    Trace(1,"[SMSInit]: SMS init started...");

    if(!SMS_SetFormat(SMS_FORMAT_TEXT,SIM0))
    {
        Trace(1,"[SMSInit]: SMS set format error");
        return;
    }
    Trace(1,"[SMSInit]: SMS set format SUCCESS");
    SMS_Parameter_t smsParam = {
        .fo = 17 ,
        .vp = 167,
        .pid= 0  ,
        .dcs= 8  ,//0:English 7bit, 4:English 8 bit, 8:Unicode 2 Bytes
    };
    if(!SMS_SetParameter(&smsParam,SIM0))
[... 182 lines cut ...]
        if(OS_WaitEvent(gpsTaskHandle, (void**)&event, OS_TIME_OUT_WAIT_FOREVER))
        {
            EventDispatch(event);
            OS_Free(event->pParam1);
            OS_Free(event->pParam2);
            OS_Free(event);
        }
    }
}

void code_Main(void)
{
    gpsTaskHandle = OS_CreateTask(
        gps_MainTask,
        NULL, 
        NULL, 
        MAIN_TASK_STACK_SIZE, 
        MAIN_TASK_PRIORITY, 
        0, 
        0, 
        "gps tracker"
    );
    OS_SetUserMainHandle(&gpsTaskHandle);
}
```

mathematikoi, 2021-09-12:
I had also problems with sms deletion, but only when trying to delete them off the flash memory
I set out to using the following procedure to delete them from sim card
`
void delete_sms_messages(void)
{
    Trace(1, "deleting sms messages from sim card");
    for (int i = 1; i <= 10; i++)
    {
        if (!SMS_DeleteMessage(i, SMS_STATUS_ALL, SMS_STORAGE_SIM_CARD))
            Trace(1, "deleting sms failed");
    }
    Trace(1, "sms messages deleted");
}`

I also set the sim card as the default storage for SMS in boot

`void sms_init(void)
{

    if (!SMS_SetNewMessageStorage(SMS_STORAGE_SIM_CARD))
    {
        Trace(1, "sms set message storage fail");
        return;
    }
}`
so far so good!

Axliyor, 2023-03-09:
> 

working 👍