Ai-Thinker-Open/GPRS_C_SDK #346: [Question] can I make it a web server?

lvdiev, 2019-06-29:
Hi every one,

I'm wondering that can I make a web server using a9g module?
Or can I make something that allows another device (let's say client) proactively connect to and send data to it?

Many thanks.

pulkin, 2019-07-06:
I do not know the answer but this is an interesting question to investigate. You should probably check whether the exposed lwip API is enough: [api_socket.h](https://github.com/Ai-Thinker-Open/GPRS_C_SDK/blob/master/include/api_socket.h) Otherwise, you should contact your cell provider regarding:

1. Using a public (static) IP address
2. Allowing incoming connections

This would be, however, quite an exotic setup. It is much easier to organize a server where all your clients and this module reports to.

lvdiev, 2019-07-25:
@pulkin thank you very much for your suggestion :)
And yes, I was researching on lwip API but it didn't work.
I created a socket server using lwip, the code compiles and runs without any error in log. But I can not connect to socket server.
Any way, as you say, building a separate server seems easier (and more elegant, I think :))
And I decided to let a9g connect to a mqtt server. Unfortunately, the out-of-the-box example does not work. I see someones posted the issues, and if you have time, please help us on that.
Thanks and happy coding!

pulkin, 2019-07-25:
I have no experience on that, sorry. You may try my [micropython port](https://github.com/pulkin/micropython) together with a [python mqtt implementation](https://randomnerdtutorials.com/micropython-mqtt-esp32-esp8266/): it should work. There is no SSL there yet but it is coming soon.

lvdiev, 2019-07-25:
Good porting! I will check it out.