Ai-Thinker-Open/GPRS_C_SDK #347: A9G won;t power ON

medida, 2019-07-08:
Hi

I'm using the SDK V2112

I have 2 A9G boards setup as i'm experimenting for a new project.
Both of them no longer respond to any power?
I can see in coolwatcher - that when i press the reset key i get 'Detected event: 0xffffffff'

but they don;t seem to be running any code or any LEDS flashing.

Any help much appreciated - either I have broken both of them! or they are stuck in some weird boot mode...
I did have them running a demo project - but stopped / unwired to insert a sim card.

Please help. 
Thanks

medida, 2019-07-08:
.... After restart everything - and sending the project to the device again - it seems its no doing something....
Very strange.

Is there any reason why the device would have LOST its coding

pulkin, 2019-09-08:
Yes, sometimes you may hang the communication. For me (linux) it happens if I start capturing events in `coolwatcher`. I just restart it.

BrickStacking, 2020-03-17:
Hi, I have the same problem as you. Now my A9G can't flash new code or active tracer. Did you solve the problem? please help !!

medida, 2020-03-17:
from memory i think i just held down the power buttons and tried to PUSH the new code at the right moment.

However i currently have a unit on my desk which is not responding to anything - i think its dead!

rahulmr8222, 2023-04-15:
@medida

rahulmr8222, 2023-04-15:
Facing problem flashing micropython into A9G,  can u help me out?

medida, 2023-04-15:
> Facing problem flashing micropython into A9G, can u help me out?

Sorry - I've not used this since 2020 as i gave up on the hardware because it was too power intensive

rahulmr8222, 2023-04-16:
Ok thankyou bro.

denicsdevices, 2023-06-20:
In my case, just changing the USB - TTL adaptor worked. My board was custom built though, and there I forgot to pull down Pin9-PWR_KEY to low with 10K resistor, so it was not booting up at first. After doing the required changes the board booted correctly but showed same behavior like o/p mentioned. I tired few different adaptors and one with FTDI Chip worked for me. CP21xx chip adaptor is also working now after flashing the A9G. BTW, make sure you are connecting the adaptor rx/tx to HST UART port and not to UART1 or 2. Also it is recommended to provide 4.2V at VBAT while flashing. Good luck

Unisolgr14, 2024-07-01:
@denicsdevices My A9G board is now not powering up (it used to work all the AT commands). Now no leds ,no AT command response. So,I bought a new TTL convertor but still the A9G board does not work.

skyelectricir, 2025-07-28:
> [@denicsdevices](https://github.com/denicsdevices) My A9G board is now not powering up (it used to work all the AT commands). Now no leds ,no AT command response. So,I bought a new TTL convertor but still the A9G board does not work.

Hello. Your problem is solved. I have the same problem :/