Ai-Thinker-Open/GPRS_C_SDK #353: Power consumtion

Duh3c, 2019-07-11:
## 1. SDK version（SDK 版本）

{
SDK V2112
}

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{
Problem appears in PM Sleep mode. And with GPS antenna plugged in.
}

Hello,
has anyone measured power consumption? 
I am developing a GPS tracker and I'm constantly getting pretty high power consumption. 
Tried to put A9G to Sleep mode and use GPS Standby mode stop and I am still getting 22mA consumption. 

In normal mode with reading GPS data around 70mA.
Is there any way to power off active GPS antenna that uses 10mA?
I am very far from the documented 1mA.
Any advice?

Regards,
Miha

updesh31, 2019-07-11:
I am getting 4ma in sleep mode with a9g. Use passive path antenna 25x25x2. Also check your power supply current at no load.

Duh3c, 2019-07-11:
Thank you for a quick answer.
You are using A9G pudding dev board right? Are you using GPS in your code?
My power consumption is 6.8mA when A9G is powered off. (GPS antenna unplugged)
If I upload a demo like gps and I put it to sleep mode before OS_Sleep, I get power consumption of around 13mA (with GPS antenna it would be 23mA). Any idea for getting it to 4mA?
Kind regards,
Miha

medida, 2019-08-05:
Don't suppose you have managed to get any further with this.

I'm looking at putting the device to sleep as well as turning on flight mode - to see if that also saves some power from the GSM modem.

K1ngM, 2019-08-05:
Sadly, no..
I tried the flight mode also, but sometimes A9G does not reconnect after turning off flight mode.
If you find the way to lower the consumption please let me know.

medida, 2019-08-06:
I using the latest version 2112 - but in the .7z download file there is no flightmode....
I have tried adding it in - but its now coming up with errors galore.

What version are you running to get it to compile with flightmode?

medida, 2019-08-07:
I've now got to the same point as you - Network_SetFlightMode(false) does not then allow the device to connect again to the network!

I've posted on their other forum to see i get a response  - http://bbs.ai-thinker.com/forum.php?mod=viewthread&tid=1452&extra=page%3D1

marimarc76, 2019-09-07:
Dear all
I've purchased an A9G pudding dev board (
https://images-na.ssl-images-amazon.com/images/I/412W7--XGgL._AC_SY400_.jpg) and I'm having the same issue in low power mode: power consumption is about 13-15 mA and NOT 3mA as in datasheet.

I'm asking if the GPS antenna included in my kit (the one in the picture) is an 'active' or 'passive' antenna. Someone could recognise it from the pic? Updesh31, your antenna is the same?

I'm asking if there are any difference about power consumption when running code compiled in 'debug' mode rather than in 'release' mode. The module drains the same current in the two modes?

Third question: the 'Trace' commands inside code could increase the power consumption? To have a lower power consumption should be better to disable the UART port for debugging?

I need to build a GPS tracker and low power consumption is the most important feature I need.

Thank you very much for your help

Marco

zkemble, 2019-09-07:
@marimarc76 What does your code look like?
Also see https://github.com/Ai-Thinker-Open/GPRS_C_SDK/issues/329#issuecomment-525875831

Pretty much all GPS antennas are active (check under the metal shield to see if they have any components), they draw around 5-10mA. They need additional circuitry to turn them off (mainly a transistor), which the pudding board does not have. The board also has a buck regulator which also consumes some power for itself.

There's no difference in power consumption between release and debug, same thing when using Trace. However, if you enable a UART port then the module will not be able to go into sleep mode, and draws around 10mA minimum. Though, it is possible to enable and disable UART ports as needed.

medida, 2019-09-07:
I'm also looking at using the device as a tracker.
I'm currently having an email conversation with ai Thinker support about been able to use the flight mode process as I need to be able to turn the modem on and off.

If / when I get a final answer from them I'll reply to these posts

marimarc76, 2019-09-07:
Thank you very very much for your help and your very fast answer! 

@zkemble: my code is quite simple and is based on few text commands (with password) to get GPS position or GSM cells in view. Next days I'll try to share my code in my account.

G think the GPS antenna provided with my pudding board kit is an active one because there are few components under the metallic plate. I'll try to get a passive one to reduce power consumption. I'll try a test with no connected antenna too. I'll let you know.

I'm asking how to disable the HST_TX HST_RX serial port to decrease power consumption. In SDK I can see that I can disable (close) only UART1 UART2 and UART_GPS. Maybe is impossible to disable HST serial port? (Otherwise it won't be possible to program the board any more) 

Finally in my code I don't need to switch GSM to flight mode because I need to read SMS immediately, anyway I think that 3 or 4 mA is a pretty good power consumption for a standby mode. With a 1200 mAh battery we can reach 12,5 days (in theory) and this is so good for my goal.

Thanks again

Marco

zkemble, 2019-09-08:
> I'm asking how to disable the HST_TX HST_RX serial port to decrease power consumption.

The HST port doesn't seem to use any additional power, don't need to worry about disabling it.

PatrykBadowski, 2021-07-30:
Hi everyone,
Wish to know that anyone figured out how to get gps tracker with low power supply and don't lose gprs connection? If so please public it somewhere :)
@marimarc76  @zkemble @medida @K1ngM @HrMitrev