Ai-Thinker-Open/GPRS_C_SDK #356: C_SDK V2112 git clone 问题后无法编译成功

xfdr0805, 2019-07-14:
## 1. SDK version（SDK 版本）

{

V2112

}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{

PS D:\SmartHome> git clone https://github.com/Ai-Thinker-Open/GPRS_C_SDK.git --recursive
Cloning into 'GPRS_C_SDK'...
remote: Enumerating objects: 4060, done.
remote: Total 4060 (delta 0), reused 0 (delta 0), pack-reused 4060
Receiving objects: 100% (4060/4060), 30.10 MiB | 2.91 MiB/s, done.
Resolving deltas: 100% (1939/1939), done.
Submodule 'libs/aliyun/iotkit-embedded' (https://github.com/aliyun/iotkit-embedded.git) registered for path 'libs/aliyun/iotkit-embedded'
Submodule 'libs/gps/minmea/src' (https://github.com/kosma/minmea.git) registered for path 'libs/gps/minmea/src'
Submodule 'libs/lvgl/src' (https://github.com/littlevgl/lvgl.git) registered for path 'libs/lvgl/src'
Submodule 'platform/csdk' (https://github.com/Ai-Thinker-Open/GPRS-C-SDK-LIB.git) registered for path 'platform/csdk'
Cloning into 'D:/SmartHome/GPRS_C_SDK/libs/aliyun/iotkit-embedded'...
remote: Enumerating objects: 181, done.
remote: Counting objects: 100% (181/181), done.
remote: Compressing objects: 100% (138/138), done.
remote: Total 32981 (delta 70), reused 99 (delta 43), pack-reused 32800
Receiving objects: 100% (32981/32981), 85.77 MiB | 4.70 MiB/s, done.
Resolving deltas: 100% (12842/12842), done.
Cloning into 'D:/SmartHome/GPRS_C_SDK/libs/gps/minmea/src'...
remote: Enumerating objects: 524, done.
remote: Total 524 (delta 0), reused 0 (delta 0), pack-reused 524
Receiving objects: 100% (524/524), 195.55 KiB | 284.00 KiB/s, done.
Resolving deltas: 100% (334/334), done.
Cloning into 'D:/SmartHome/GPRS_C_SDK/libs/lvgl/src'...
remote: Enumerating objects: 4, done.
remote: Counting objects: 100% (4/4), done.
remote: Compressing objects: 100% (4/4), done.
remote: Total 19720 (delta 0), reused 2 (delta 0), pack-reused 19716
Receiving objects: 100% (19720/19720), 14.30 MiB | 1.97 MiB/s, done.
Resolving deltas: 100% (15647/15647), done.
Cloning into 'D:/SmartHome/GPRS_C_SDK/platform/csdk'...
remote: Enumerating objects: 344, done.
remote: Total 344 (delta 0), reused 0 (delta 0), pack-reused 344
Receiving objects: 100% (344/344), 170.61 MiB | 6.39 MiB/s, done.
Resolving deltas: 100% (154/154), done.
error: Server does not allow request for unadvertised object e6bd3556039538ac504c005f0c31dbdd0e906658
Fetched in submodule path 'libs/aliyun/iotkit-embedded', but it did not contain e6bd3556039538ac504c005f0c31dbdd0e906658. Direct fetching of that commit failed.
PS D:\SmartHome> cd .\GPRS_C_SDK\

PS D:\SmartHome\GPRS_C_SDK> ./build.bat demo mqtt
CSDTK PATH: D:\SmartHome\CSDTK42
number of processors: 8
D:/SmartHome/GPRS_C_SDK/platform/compilation/cust_rules.mk:8: D:/SmartHome/GPRS_C_SDK/platform/csdk/memd.def: No such file or directory
D:/SmartHome/GPRS_C_SDK/platform/compilation/cust_rules.mk:115: WARNING: No platform elf file at path: D:/SmartHome/GPRS_C_SDK/platform/csdk/debug
D:/SmartHome/GPRS_C_SDK/platform/compilation/cust_rules.mk:128: WARNING: No platform lod file at path: D:/SmartHome/GPRS_C_SDK/platform/csdk/debug
-- System Version : WINDOWS
---------------------------------------------------
D:/SmartHome/GPRS_C_SDK/platform/compilation/cust_rules.mk:8: D:/SmartHome/GPRS_C_SDK/platform/csdk/memd.def: No such file or directory
D:/SmartHome/GPRS_C_SDK/platform/compilation/cust_rules.mk:8: D:/SmartHome/GPRS_C_SDK/platform/csdk/memd.def: No such file or directory

[MAKE]  init
D:/SmartHome/GPRS_C_SDK/platform/compilation/cust_rules.mk:8: D:/SmartHome/GPRS_C_SDK/platform/csdk/memd.def: No such file or directory

D:/SmartHome/GPRS_C_SDK/platform/compilation/cust_rules.mk:8: D:/SmartHome/GPRS_C_SDK/platform/csdk/memd.def: No such file or directory
[CC]   sdk_init.o                  <== sdk_init.c
[AR]   libinit_debug.a

[MAKE]  libs
D:/SmartHome/GPRS_C_SDK/platform/compilation/cust_rules.mk:8: D:/SmartHome/GPRS_C_SDK/platform/csdk/memd.def: No such file or directory

[MAKE]  libs/gps
D:/SmartHome/GPRS_C_SDK/platform/compilation/cust_rules.mk:8: D:/SmartHome/GPRS_C_SDK/platform/csdk/memd.def: No such file or directory

[MAKE]  libs/gps/minmea
D:/SmartHome/GPRS_C_SDK/platform/compilation/cust_rules.mk:8: D:/SmartHome/GPRS_C_SDK/platform/csdk/memd.def: No such file or directory

D:/SmartHome/GPRS_C_SDK/platform/compilation/cust_rules.mk:8: D:/SmartHome/GPRS_C_SDK/platform/csdk/memd.def: No such file or directory
[CPP]  %-27s <== %s\n minmea.asm minmea.S
mips-elf-gcc: src/minmea.S: No such file or directory
mips-elf-gcc: no input files
D:/SmartHome/GPRS_C_SDK/platform/compilation/cust_rules.mk:997: recipe for target 'D:/SmartHome/GPRS_C_SDK/build/mqtt/libs/gps/minmea/obj/debug/minmea.o' failed
make[6]: *** [D:/SmartHome/GPRS_C_SDK/build/mqtt/libs/gps/minmea/obj/debug/minmea.o] Error 1
D:/SmartHome/GPRS_C_SDK/platform/compilation/cust_rules.mk:834: recipe for target 'D:/SmartHome/GPRS_C_SDK/build/mqtt/libs/gps/minmea/lib/libminmea_debug.a' failed
make[5]: *** [D:/SmartHome/GPRS_C_SDK/build/mqtt/libs/gps/minmea/lib/libminmea_debug.a] Error 2
D:/SmartHome/GPRS_C_SDK/platform/compilation/cust_rules.mk:1030: recipe for target 'dependencies' failed
make[4]: *** [dependencies] Error 2
D:/SmartHome/GPRS_C_SDK/platform/compilation/cust_rules.mk:1030: recipe for target 'dependencies' failed
make[3]: *** [dependencies] Error 2
D:/SmartHome/GPRS_C_SDK/platform/compilation/cust_rules.mk:1030: recipe for target 'dependencies' failed
make[2]: *** [dependencies] Error 2
D:/SmartHome/GPRS_C_SDK/platform/compilation/cust_rules.mk:748: recipe for target 'D:/SmartHome/GPRS_C_SDK/build/mqtt//mqtt.srec' failed
make[1]: *** [D:/SmartHome/GPRS_C_SDK/build/mqtt//mqtt.srec] Error 2
D:/SmartHome/GPRS_C_SDK/platform/compilation/cust_rules.mk:395: recipe for target 'lod' failed
make: *** [lod] Error 2
grep: platform\csdk\memd.def: No such file or directory
grep: platform\csdk\memd.def: No such file or directory
grep: build\mqtt\mqtt.map: No such file or directory
grep: build\mqtt\mqtt.map: No such file or directory
grep: build\mqtt\mqtt.map: No such file or directory
找不到操作数。
找不到操作数。
-------------------------------------------------
ROM    total: Bytes     used: Bytes
RAM    total: Bytes     used: Bytes
=================================================
Start Time : 10:03:03.49
End   Time : 10:03:07.32
Build Time : 3.983s
=================================================
PS D:\SmartHome\GPRS_C_SDK>

}

---

xuhongv, 2019-08-03:
请问你解决了吗  我也是遇到这问题

medida, 2019-08-07:
I found that if you download the additional github respositories into the correct folders it them works - but you have to do it manually

these are all the git modules you need to do 

[submodule "platform/csdk"]
	path = platform/csdk
	url = https://github.com/Ai-Thinker-Open/GPRS-C-SDK-LIB.git

[submodule "libs/gps/minmea/src"]
	path = libs/gps/minmea/src
	url = https://github.com/kosma/minmea.git
[submodule "libs/aliyun/iotkit-embedded"]
	path = libs/aliyun/iotkit-embedded
	url = https://github.com/aliyun/iotkit-embedded.git
[submodule "libs/lvgl/src"]
	path = libs/lvgl/src
	url = https://github.com/littlevgl/lvgl.git