Ai-Thinker-Open/GPRS_C_SDK #361: Trying to use cJSON

medida, 2019-07-22:
SDK v2112

I'm tring to get cJSON to work with my pudding a9G board.

I have downloaed the .c and .h files from [https://github.com/DaveGamble/cJSON](url)

I have added the files to the libs folder - but i'm not able to compile the code any longer. 

Example of the code i'm trying run - with out this it works fine
`cJSON *root =   cJSON_Parse(reply_buffer);
    char *ack = cJSON_GetObjectItem(root, "ack")->valuestring;

    Trace(2, "@@@ ack: %s", ack);
`
I get an error of:
 error: limits.h: No such file or directory

But when i add the limits.h file - i get more errors due to the 
error: _mingw.h: No such file or directory

and lots of other types being redefined.

Any help please - This is my first time with C.....

medida, 2019-07-25:
Using the latest version from github finally got it working

closed 2019-07-25 by medida

xuhongv, 2019-08-03:
- hi when i clone it ,but error  to show ,please tell me how you to do

` git clone https://github.com/Ai-Thinker-Open/GPRS_C_SDK.git --recursive
`

medida, 2019-08-03:
I downloaded the code from this link: https://github.com/Ai-Thinker-Open/GPRS_C_SDK/releases/tag/V2.112

This version has a cJSON example it which worked.