Ai-Thinker-Open/GPRS_C_SDK #364: PM_Sleep - is it possible to shutdown for X seconds

medida, 2019-07-25:
Hi

So i'm looking at saving battery while using the A(G board

I'm currently testing the PM_Sleep - but i want to know if there is any way to reduce power any more....

Ideally shutdown and restart in X Seconds....

It would be nice if there was some answers to questions posted - This is seems to be nice little product but unless i can get guidance then its not going to be OK moving forward..
Is there anywhere else i can get advice from about this product where there is a more active community?

kjfdebruin, 2019-08-29:
I would be interested in this. Did you manage to find out anything?

medida, 2019-08-29:
@kjfdebruin i've had no reply from the company but i have had various replies on other posts
these post have been usefull

[https://github.com/Ai-Thinker-Open/GPRS_C_SDK/issues/333#issuecomment-526040778](url)

[https://github.com/Ai-Thinker-Open/GPRS_C_SDK/issues/329#issuecomment-525939185](url)

At the moment i am testing with PM_Sleepmode - but i'm going to try shutting down the GPIO as well this week.

I have also investigated the idea of the microcontroller / p-mosfet which would shut down the entire device until the next time interval - but the ones i have found only have a maximum of 7200 seconds.

Ideally i only want the device to wake up once a day!

kjfdebruin, 2019-08-29:
Thanks so much - we want to do exactly the same thing! I'll post on here if I have any success!

medida, 2019-08-29:
there is also some activity on there other forum....
[http://bbs.ai-thinker.com/forum.php?mod=viewthread&tid=1221&extra=page%3D1](url)

pulkin, 2019-09-08:
Just set up the watchdog timer before shutting down your module. Once the watchdog triggers, the module will start up.

medida, 2019-09-08:
I've not seen the watchdog timer. Will that wake a unit when it's been powered off?

pulkin, 2019-09-08:
It *should* work like that. The problem is that I was never able to shut down module (it just restarts in my case), at least while it was USB-powered, so I was never able to test whether it wakes up after. The watchdog alone, however, works.

Pasboe, 2019-12-22:
The low power for me is to high I going to try and use an I2C Realtime Clock IC  like a DS3231 with alarm output and very low power usage. This way I can Switch off the A9G completely and wake it up at a programmed date/time.

medida, 2019-12-22:
Would you also use a mosfet

Pasboe, 2019-12-23:
@medida  Depends on the circuit still doing some research :)

medida, 2019-12-23:
Hi
It seem Ai-Thinker may have done this before.... i'm currently having an email conversation with them and this is their latest reply.

Yes, you can add an extra microcontroller + MOSFET to manager the power to the device.

Our customer do the same thing, but they choose their microcontroller + MOSFET. Do you have your own plan?

Now waiting for them to provide some details - i'll report back when i get further info

medida, 2019-12-23:
Ai-Thinkers reply....................
Sorry I don't know more details, one of our customer is using STM series microcontroller.
The model name is STM8L152, hope it's helpful.
 
I currently have an attiny85 and mosfet on order so hoping to make something usable...

punee995, 2019-12-23:
It Shutdown time is between 100ms and 2hour then I recommend to look at [TLP5010](http://www.ti.com/lit/ds/symlink/tpl5010.pdf).

medida, 2020-01-28:
@Pasboe have you got any further with using the RTC.

I'm currently trying to get a DS3231 to return the correct time after its been set, but it just won't work.

Any advice greatly appreciated.

Pasboe, 2020-01-29:
@medida Sorry I'm looking at a smaller RTC clock like an RX8130.
My goal is to design a small as possible PCB.
For this moment I'm only work my way trough the examples to learn.
I have the A9G running with GPS, GPRS, SMS and GPIO.  
Zkemble posted his great project https://github.com/Ai-Thinker-Open/GPRS_C_SDK/issues/396 this helps a lot.  I hope the SDK is going to be updated soon and hopefully some bug's solved.

medida, 2020-01-29:
@Pasboe thanks for that - I'm currently stuck sending commands to the RTC via the i2c commands.

Nearly all I2C command examples are arduino Wire based - which does not help as the A9G process can take an array of data - but i don't know if thats the correct way to do it.

I can send data without an error - but the timestamp from the RTC does not change.

I can read the datetime stamp from the RTC ( although its not correct ) 
for example using......
I2C_ReadMem(I2C_ACC, a, 0x00, 1, &pdata[0], 7, t);
uint8_t second = pdata[0];

but if i try to update an element of the time it just won;t work

  I2C_WriteMem(I2C_ACC, a, 0x06, 1, decToBcd(2020), 1, t);

bcdToDec - is a function to convert the value into the correct format ( i think ) 
uint8_t bcdToDec(uint8_t val)
{
    // Convert binary coded decimal to normal decimal numbers
    return ((val / 16 * 10) + (val % 16));
}

 second = (second >> 4) * 10 + (second & 0x0F);

Pasboe, 2020-01-29:
@medida. I wil start testing I2C soon let you know if I can get it to work.

Pasboe, 2020-01-29:
@medida  I2C_WriteMem(I2C_ACC, a, 0x06, 1, decToBcd(2020), 1, t);
                                                                     ^ should this value not be 4?
You are sending 4 bytes not 1 !

medida, 2020-01-31:
@Pasboe the code from https://github.com/zkemble/MailboxNotifier/blob/master/firmware/a9g/src/bme280.c helped me get it working better. I'll post the full test code in a day or so ( although a different RTC chip - it may be useful) 

I can now set the time and alarms - just need the alarm trigger to fire the right output to then demonstrate its actually triggering something ( e.g. LED )

Pasboe, 2020-01-31:
@medida Sounds all good looking forward to you're code.

Pasboe, 2020-02-07:
I've managed to read and write to an I2C Eprom. 
But it took me to long to figure everything out. 
Loaded the AT firmware back in to the A9G and hooked op an ATMEGA328.
Use a mosfet to switch power on/off and if I set the ATMega in low power my current drops to about 0,22uA. Thats fine for me. Now I can store settings into the EEPROM of the ATmega. 
Only a few hours of programming and I'm receiving and sending SMS messages. Go to low power and wake up by a trigger from my RTC or button.  Now designing a PCB.

medida, 2020-02-07:
Could I get a pic of your setup. Not sure I follow.

So your using the atmega to control the rtc, but waking the pudding to send sms etc..

Pasboe, 2020-02-07:
#medida With the default firmware the A9G can be managed with AT commands. This way I turn it into a slave device. Instead running my own code in the A9C I'm putting my code into a ATMega328. I'm using the arduino IDE to write my program and this way have access to a lot of libraries. The only connection between the A9G and ATMega is a serial connection. The idea came from this board: https://www.makerfabs.com/maduino-zero-a9g.html.
Except this board is to big and can't handle low power mode.
Can't upload my schematic here sorry,

medida, 2020-02-07:
Hi. Thanks for that.

camp-easy, 2024-04-06:
hello, can anyone tell me how to shut down the A9G? it is possible to do with the powerkey or it just restart? thanks in advance:)