Ai-Thinker-Open/GPRS_C_SDK #368: aws iot connect problem

dinesh12jain, 2019-08-03:
## 1. SDK version（SDK 版本）

{

CSDTK42_Update2_18_7_27

}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{

I want to connect aws iot mqtt broker with A9 device
i use demo mqtt_ssl code

i have tryied many times but always it give me errors
like 258 or 256 error
[image]

}

---

malte777, 2020-05-17:
Hi, I also have this problem. Did you solve it?

malte777, 2020-05-18:
I have it running now, by making these changes:
change
    ci.ssl_min_version   = MQTT_SSL_VERSION_SSLv3;
to
    ci.ssl_min_version   = MQTT_SSL_VERSION_TLSv1_2;

delete:
    ci.entropy_custom    = "GPRS_A9";
    ci.will_qos = 2;
    ci.will_topic = "will";
    ci.will_retain = 1;
    memcpy(strstr(willMsg,"GPRS")+5,imei,15);
    ci.will_msg = willMsg;

Muhan1995, 2020-12-20:
@farmsensor  Hello! I have this problem too! Deleted as you indicated. Here is some of the code
MQTT_Client_t* client = MQTT_ClientNew();
MQTT_Connect_Info_t ci;
MQTT_Error_t err;
memset(&ci,0,sizeof(MQTT_Connect_Info_t));
ci.client_id = imei;
ci.client_user = CLIENT_USER;
ci.client_pass = CLIENT_PASS;
ci.keep_alive = 60;
ci.clean_session = 1;
ci.use_ssl = true;
ci.ssl_verify_mode = MQTT_SSL_VERIFY_MODE_REQUIRED;
ci.ca_cert = ca_crt;
ci.ca_crl = NULL;
ci.client_cert = client_crt;
ci.client_key  = client_key;
ci.client_key_passwd = NULL;
ci.broker_hostname = BROKER_HOSTNAME;
ci.ssl_min_version = MQTT_SSL_VERSION_TLSv1_2;
ci.ssl_max_version = MQTT_SSL_VERSION_TLSv1_2;

It connects to the server and immediately disconnects. Help, how you defeated the problem in more detail. 
BROKER_IP  "a19eh4w67cred3-ats.iot.us-west-2.amazonaws.com"
BROKER_PORT 8883
BROKER_HOSTNAME "us-west-2.amazonaws.com"
CLIENT_USER NULL
CLIENT_PASS NULL
SUBSCRIBE_TOPIC "$aws/things/ThingOne/shadow/update/accepted"
PUBLISH_TOPIC   "$aws/things/ThingOne/shadow/update"
PUBLISH_INTERVAL 5000
PUBLISH_PAYLOEAD "{ \"state\": { \"reported\": { \"state\": 0 } }}"

ramitdour, 2022-01-15:
> I have it running now, by making these changes: change ci.ssl_min_version = MQTT_SSL_VERSION_SSLv3; to ci.ssl_min_version = MQTT_SSL_VERSION_TLSv1_2;
>
> delete: ci.entropy_custom = "GPRS_A9"; [...]

this is still not working

I am getting these two errors : 
Missing encoded blocks 0/1 NbTx 2 TxBitMap 0x0\n
pending TBF,invalid data

Has anyone successfully been able to connect A9/A9G to AWS IoT core?