Ai-Thinker-Open/GPRS_C_SDK #372: I wonder this module can steal my data?

txhai, 2019-08-19:
Hello, I am trying to work with the SDK but I cannot find the implementation of API functions. I think if there are hidden codes that send the GPS data to a certain server? 

How do I know this module does not do that?

Thanks.

pulkin, 2019-09-08:
There is no way to do that. GPS, however, is a separate chip via UART so it is easy to take control of this kind of communication. There is base station tracking which you cannot take control of.

txhai, 2019-09-09:
Thank you for your answer @pulkin, I really appreciate it. But can you describe in more details? I am new to IoT and cannot understand all of your words. I think there are instructions that `Start GPS Chip -> Get the data -> Start GPRS -> Send via TCP`.

pulkin, 2019-09-09:
GPS tracking example: https://github.com/Ai-Thinker-Open/GPRS_C_SDK/tree/master/demo/gps_tracker/src

txhai, 2019-09-09:
Edited: I mean the SDK contains hidden instructions would do steal the data, like your file attached above

pulkin, 2019-09-09:
Yes, this is possible.

txhai, 2019-09-09:
Thank you! This is a great module but give the doubt.

closed 2019-09-09 by txhai