Ai-Thinker-Open/GPRS_C_SDK #378: AT Commands

medida, 2019-09-17:
I'm trying to turn OFF / ON the GSM module to save power and to make the unit as dormant as possible.

AI-Thinker have responded saying i can send AT commands to  turn flight mode on / OFF

Turn on the flight mode commands: at+cfun=0
Turn off the flight mode commands: at+cfun=0

But From the SDK - how can i send AT commands?

I've found there AT commands firmware: https://github.com/Ai-Thinker-Open/GPRS-AT
but i would like to use the current SDK and just poke AT commands as required.

Any suggestions?

giuspe01, 2019-12-05:
hello, did you find a solution? I'm in the same situation!!

medida, 2019-12-05:
Hi

Yes - ish.

Have a look at [https://github.com/Ai-Thinker-Open/GPRS_C_SDK/issues/333](url)

And this one [https://github.com/Ai-Thinker-Open/GPRS_C_SDK/issues/258#issuecomment-558205523](url)

You cannot call AT commands from their SDK - but i did find away of starting the GSM successfully after it had been in flightmode

punee995, 2019-12-23:
Have you tried this??
https://github.com/Ai-Thinker-Open/GPRS_C_SDK/blob/91832c812f2f285b46f723ef8dfb1b61a8e8661d/include/api_network.h#L92

medida, 2019-12-23:
Thabks...

I managed to turn flight mode off and reconnect ,

closed 2019-12-24 by medida