Ai-Thinker-Open/GPRS_C_SDK #379: AGPS and BRDC GPD file format?

jupyterknight, 2019-09-19:
I want to use AGPS feature in this module, I see it download a file named `brdcGPD.dat`.

https://github.com/Ai-Thinker-Open/GPRS_C_SDK/blob/91832c812f2f285b46f723ef8dfb1b61a8e8661d/libs/gps/include/gps.h#L30-L32

Does anyone know what is this file and what is its format? And any idea I can make this file for my country?

Thank you very much.

P/S: It looks like the company no longer support this SDK, but I hope that there is someone know this.

pulkin, 2019-11-04:
Hi, I am working on a micropython [port](https://github.com/pulkin/micropython/tree/master/ports/gprs_a9) where you can poll base stations and look up their location.

win32cpp, 2019-11-26:
I have a similar question above.
Does anyone have any information on creating a BRDC GPD file?

pulkin, 2019-11-27:
I implemented `agps` in my [port](https://github.com/pulkin/micropython/tree/master/ports/gprs_a9#misc). It is possible to use a local database but they are quite large so the performance will be really low.

aerosales10, 2020-03-27:
I'm also trying to use the AGPS module but it fails:

- [GPS][src/gps.c][Http_Get][786]>>"recv len:3577,data:HTTP/1.1 200 OK Content-Type: application/x-zip-compressed
- [GPS][src/gps.c][GPS_AGPS][908]>>"GPD file length:3332"
- [GPS][src/gps.c][GPS_AGPS][914]>>"GPD file length(with padding 0):3584"
- [GPS][src/gps.c][GPS_CMDSend][378]>>"gps send nmea cmd:$PGKC149,1,9600*1D,len:20"
- [GPS][src/gps.c][OnCmdAckFail][404]>>"gps ack time out"
- [GPS][src/gps.c][GPS_SendCMDWaitAck][425]>>"GPS exec command fail"
- [GPS][src/gps.c][GPS_SendWaiteNormalAck][468]>>"ack cmd check fail, wish:1, actual:0"
- [GPS][src/gps.c][GPS_AGPS][921]>>"set binary mode fail"
- AGPS RESULT:0

win32cpp, 2023-08-25:
Has anyone identified the GPD file structure yet?