Ai-Thinker-Open/GPRS_C_SDK #382: How to use this API_EVENT_ID_POWER_INFO

stbungaof, 2019-10-04:
//power
    API_EVENT_ID_POWER_INFO                        , //param1: (PM_Charger_State_t<<16|charge_level(%)) , param2: (PM_Battery_State_t<<16|battery_voltage(mV))

form [https://github.com/Ai-Thinker-Open/GPRS_C_SDK/blob/master/include/api_inc/api_event.h](url)

stbungaof, 2019-10-04:
How to get data form param1 and param2

allejp, 2020-04-16:
Hi, you can extract info in this way:

```
        uint8_t percentage = 0xFFFF & pEvent->param1;
        uint16_t voltage = 0xFFFF & pEvent->param2;

        PM_Charger_State_t cs = 0xFFFF & (pEvent->param1 >> 16);
        PM_Battery_State_t bs = 0xFFFF & (pEvent->param2 >> 16);
```

I noticed that the values are not updated quickly, the most updated value is obtained in this way:
```
        uint8_t percentage = 0;
        uint16_t voltage = PM_Voltage(&percent);
```