Ai-Thinker-Open/GPRS_C_SDK #388: sqrt

reallyczx, 2019-11-22:
## 1. SDK version（SDK 版本）

{

}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{

}

---

Johhnzero, 2019-11-22:
？？

iz1gww, 2020-01-13:
I have the same problem...
if i use sqrt function with a number (int or float) es: 

float tmp = sqrt(2.3);

 the compiler do the job without problem... but if i pass a variable 

float numb = 2.3;
float tmp = sqrt(numb);

the compiler bring up an error : "undefined exception to 'sqrt'"

I have download and installed the lastest GPRS_C_SDK and CSDTK

thank you very mutch
Alex

iz1gww, 2020-01-14:
> I have the same problem...
> if i use sqrt function with a number (int or float) es:
> 
> float tmp = sqrt(2.3);
> 
> the compiler do the job without problem... but if i pass a variable
> 
> float numb = 2.3;
> float tmp = sqrt(numb);
> 
> the compiler bring up an error : "undefined reference to 'sqrt'" and "relocation truncated to fit: R_MIPS16_26 against 'sqrt'"
> 
> I have download and installed the lastest GPRS_C_SDK and CSDTK
> 
> thank you very mutch
> Alex

resolved with a total reinstallation of sdk chaintools...
thanks 
Alex