Ai-Thinker-Open/GPRS_C_SDK #389: FastPf::FlashCommandUnacknowledged

AfroJackal, 2019-11-28:
I have been consistently encountering the shown below for several days now. I have looked at solutions provided on similar issues and tried implementing the to no success. Does any one have an idea of why this happens and a possible solution
[image]

izobara, 2020-01-07:
Same for me. Pls help! 
[image]

izobara, 2020-01-08:
@AfroJackal Have you finish your flashing with success? Maybe you have a solution now?

izobara, 2020-01-08:
In my case it was an issue with CHG340 USB-UART converter. Changed to CP and everything works well.