Ai-Thinker-Open/GPRS_C_SDK #392: Unable to download any on pundit a9g

dhernandezl, 2019-12-22:
I reopen the user's question  bitwarriors, I have the same problem could you help me.
I leave the previous user's question because it is the same problem.

Goodmorning,
after some little problems on making the lod files, now (around 2 weeks) I'm not able to download the new firmware. My coolwatcher (2.00.02) stops on this error, after a workig first connection:

`CHST_Read32': Request to coolhost timed out ! (CoolHostTimeoutError)

./rbbase/common/coolhost_connections.rb 161 CHST_Read32' ./rbbase/common/coolhost_connections.rb 161 read32'
./rbbase/common/base.rb 85 read' ./rbbase/common/base.rb 239 read'
./chipgen/Modem2G/toolpool/scripts/8809/target_executor.rb 72 enterHostMonitor' ./chipgen/Modem2G/toolpool/scripts/ChipStd/target_executor.rb 366 session'
./chipgen/Modem2G/toolpool/scripts/ChipStd/ramrun.rb 12 doramrunXfer' ./chipgen/Modem2G/toolpool/scripts/ChipStd/ramrun.rb 62 ramrunXfer'
./chipgen/Modem2G/toolpool/plugins/fastpf/fastpf.rb 212 dofastpfXfer' ./chipgen/Modem2G/toolpool/plugins/fastpf/fastpf.rb 718 dofastpf'
./chipgen/Modem2G/toolpool/plugins/fastpf/fastpf.rb 788 fastpf' ./chipgen/Modem2G/toolpool/plugins/fastpf/fastpfplugin.rb 83 fastpfGenericGo'
./chipgen/Modem2G/toolpool/plugins/fastpf/fastpfplugin.rb 112 `fastpfGo'

the CSDK is the 4.2 and the GPRS_SDK is the latest on line, the usb-uart is the cp2102 and the tx->hst-rx and rx->hst-tx pins are connected.
I'm very frustrated because no way to get rid of this error.
Any helps will be gratefully accepted +1
Thank you

l-sannin, 2020-01-02:
Hello I have the same problem, on i2c bus, blocking  i think.
I need to power down the module and change the ubs port on my computer, and works again, hope this help.

marcosasilvalepe, 2020-12-03:
Same issue here. 

The module simply froze for one minute to the next and no way revive it. I can't communicate with the module at all.

[image]

Still no solution ??

lamqhoang, 2020-12-04:
I have solved the issue by flashing the Factory AT command binary. 
https://docs.ai-thinker.com/en/gprs/firmware

marcosasilvalepe, 2020-12-07:
It worked!! Thanks!!