Ai-Thinker-Open/GPRS_C_SDK #393: FOTA Cannot update Firmware.

MaxBoyman, 2019-12-22:
## 1. SDK version（SDK 版本）

{
V2.112
}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{
whats the problem with FOTA i exactly did what it said in the FOTA section of document it doesn't work.

[image]

what should i do with this.
i will say what i got from the documentation briefly:
1- cleaning hex folder by "./build.bat clean"
2- compiling code with "./build.bat demo fota" without considering that if it is a realese or debug version
3- back up the "hex/fota" folder somewhere else and copy "fota_debug_ota.lod" to "GPRS_CSDK/" then rename it to old.lod
4- changing whatever you want to provide after program update. for example changing version by modifying "SOFT_VERSION" from "V3.0" to "V3.1".
5-cleaning "hex/fota" directory and compiling again without considering that if it is a realese or debug version
6-copy "fota_debug_ota.lod" to "GPRS_CSDK/" and rename it to new.lod
7- executing batch with this code: "./build.bat fota old.lod new.lod V3.0tonew.pack"
8- uploading "V3.0tonew.pack" to the host you provided it's link in code.
9- uploading code to "A9" or "A9G" and waiting for update.
I did this procedure with no success. First i used an https server but then i changed the server security to http. again with no success.
}

---

TurboElectric, 2024-01-29:
Hello. Did you manage to solve the problem? I have the same situation.

MaxBoyman, 2024-02-05:
No i didn't, and because of this and other bugs and problems and also because the developer didn't answer any questions, i decided to change the hardware. And I advice you too. I was working for 6 months on this module and then i needed to change the hardware because of the lack of software support and various bugs and problems.