Ai-Thinker-Open/GPRS_C_SDK #396: Full project using the A9G - mail notifier (not an issue, but could help with project design)

zkemble, 2020-01-08:
This isn't an issue, but it could help people with their firmware and hardware designs for the A9/A9G. I've made a mail notifier using the A9G that sends JSON data to a web server when triggered. It also has a BME280 which communicates with the module via I2C, uses UART to communicate with an ATtiny402 microcontroller and can last for around 4-5 years on a single charge of a 10440 lithium battery (~320mAh). It also has a GPS tracker mode, and has hardware for controlling power to an active GPS antenna.

https://github.com/zkemble/MailboxNotifier

https://blog.zakkemble.net/remote-mail-notifier-and-gps-tracker/

pulkin, 2020-01-26:
Looks decent! You could try doing the same with my micropython [port](https://github.com/pulkin/micropython/tree/master/ports/gprs_a9).

Pasboe, 2020-01-29:
Great project thanks for sharing.  
Hope to learn from it.
Thanks :)

medida, 2020-01-29:
this looks great.
when you were using the I2C from the A9G did you have to send all data i byte at a time or could you send through an array of items...

I'm trying to use I2C to communicate to a RTC module and althoguh i can request data i cannot set the current datetime on it

zkemble, 2020-01-29:
@medida I used I2C_Transmit() and I2C_Receive() to transfer multiple bytes at a time, see https://github.com/zkemble/MailboxNotifier/blob/master/firmware/a9g/src/bme280.c

medida, 2020-01-31:
@zkemble thanks - the I2C Transmit is much better code than the demo code from Ai-Thinker..
Not only was my code not functioning fully - a connection on my breadboard was faulty :(

Pasboe, 2020-02-07:
Got I2C working thanks to @zkemble but it took me to long to learn how to use the API's. 
Loaded the AT firmware and hooked up a ATMega328 (TQFP body). 
Now after a few hours programming I have low power mode (0,22uA), triggers, SMS, RTC and store settings into EEPROM. I'm happy now :)

medida, 2020-02-07:
Sorry for sending the code @Pasboe been busy busy week.

So are you using an atmega to control the rtc which then wakes the pudding?

AhmadSySy, 2021-03-13:
@zkemble Do you have a simple code about connect and get temperature data from the **Adafruit_MLX90614** temperature sensor with the **A9G development board** 
Can you help me please?