Ai-Thinker-Open/GPRS_C_SDK #401: MQTT demo

benqll, 2020-02-11:
## 1. SDK version（SDK 版本）

{

}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{
The mqll_ssl demo cannot work.  It just continuous show the CSQ message. What is the meaning of the CSQ? 

[image]

}

---

zkemble, 2020-02-11:
CSQ is the signal strength, where 0 is the lowest and 31 is the highest. 14
- 16 is good. Since you haven't got a "network activate success..." message
you've probably not set the correct APN settings (line 175), which would be
why you're not seeing any MQTT stuff.

benqll, 2020-02-11:
@zkemble 
Thanks for replying. I have tried setting the APN but it seems not work for me. It just keep showing the CSQ and do nothing. Sometimes it even didnt show network register and attach. I have tried another sim card but nothing help. Both sim card can access the internet in the my phone. 

[image]

Soram-jxx, 2020-03-08:
> CSQ is the signal strength, where 0 is the lowest and 31 is the highest. 14 - 16 is good. Since you haven't got a "network activate success..." message you've probably not set the correct APN settings (line 175), which would be why you're not seeing any MQTT stuff.

When I tested demo/ MQTT, the MQTT connection failed and the error status was “MQTT_CONNECTION_REFUSED_PROTOCOL_VERSION ”.Why did this error occur?

Soram-jxx, 2020-03-08:
When I tested demo/ MQTT, the MQTT connection failed and the error status was “MQTT_CONNECTION_REFUSED_PROTOCOL_VERSION ”.Why did this error occur?

@zkemble

zkemble, 2020-03-08:
I've never used MQTT, but I would assume that whatever you're trying to connect to doesn't like the MQTT protocol version that the A9G uses. It seems v3.1.1 and v5 are the main ones.