Ai-Thinker-Open/GPRS_C_SDK #402: Developer wanted (Not Issue)

procopio, 2020-02-11:
Hi, 

I'm looking for a developer with some experience in A9G Pudding to adapt the demo GPS Tracker demo to our needs.
(https://github.com/Ai-Thinker-Open/GPRS_C_SDK/blob/master/demo/gps_tracker/src/demo_gps_tracker.c) 

Basically the changes will be to store the data in a different format and post a different format http post to our server.

We already have a tracker working with Arduino + GPS + GSM, but if it's possible to do with  A9G Pudding it will be smaller and easier to assemble.

Please write privately to me saying if:
Do you have any interest in developing this project?
Do you have experience doing this with A9G Pudding?

We will send you more details about our needs to you prepare a proposal.

Thanks,
Joao

Maerkert, 2020-02-12:
Hey Joao, check out my A9G project. I am currently developing a GPS tracking app for cars with a custom Python TCP backend and an Android App to show the location. It may help you. Contact me if it looks good for you. Greetings Fabian

pulkin, 2020-02-16:
I do not want to ruin this party but [the GPS tracker example](https://github.com/pulkin/micropython/blob/master/ports/gprs_a9/examples/example_50_mqtt.py) in my micropython port is 32 lines of code.

Let me say it once again: **32 lines of code** which includes a generous comment in the header and downloading all the needed packages. You might be able to adapt it to your needs within minutes. But if you **really** looking for help please post all the details publicly here.

Maerkert, 2020-02-18:
Nice and simple code you got there. Did not knew that micropython supports the A9G. But yout code does not satisfy my needs. For Joao it could be benificial.

pulkin, 2020-02-18:
If you share your needs I will consider implementing them in short or long term.

procopio, 2020-02-18:
Thank you very much guys! 
I found a developer to do it.

closed 2020-02-18 by procopio