Ai-Thinker-Open/GPRS_C_SDK #404: Open and Close GPS, error "Uart not enabled"

allejp, 2020-02-15:
## 1. SDK version（SDK 版本）

{

C_SDK 2108

}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{

I'm trying to manage GPS by MQTT message, actually I'm using the MQTT demo with two new events in MqttTaskEventDispatch task.

I can Open the GPS the first time and Close it. but when i try to re-GPS_Open(NULL) the program crash whit this error:

> Assert received!! Reading detail...
> ASSERT DETAIL : 
> Usart not enabled

So:

1. GPS_Open(NULL) <-- OK
2. GPS_Close() <-- OK
3. GPS_Open(NULL) <-- FAIL

This is my code

```
void MqttTaskEventDispatch(MQTT_Event_t *pEvent)
{
    switch (pEvent->id)
    {
    case MQTT_EVENT_CONNECTED:
        GPS_Init();
        mqttStatus = MQTT_STATUS_CONNECTED;

        // Subscribe to some topics...

        StartTimerPublish(gpsInterval, pEvent->client);
        break;
    case MQTT_EVENT_DISCONNECTED:
        mqttStatus = MQTT_STATUS_DISCONNECTED;
        GPS_Close();
        break;
    case GPS_OPEN:
        Trace(1, "Open GPS");
        
       // GPS_Init();
        GPS_Open(NULL);

        GPS_Info_t *gpsInfo = Gps_GetInfo();
        //wait for gps start up, or gps will not response command
        while (gpsInfo->rmc.latitude.value == 0)
            OS_Sleep(1000);

        uint8_t buffer[150];
        if (!GPS_GetVersion(buffer, 150))
            Trace(1, "get gps firmware version fail");
        else
            Trace(1, "gps firmware version:%s", buffer);

        if (!GPS_SetFixMode(GPS_FIX_MODE_LOW_SPEED))
            Trace(1, "set fix mode fail");
        break;
    case GPS_CLOSE:
        Trace(1, "Close GPS");
        GPS_Close();
        break;
    default:
        break;
    }
}
```

I tried also version SDK 2112 but int this case the program crash as soon as this event was received (related to mqtt demo):
```
case API_EVENT_ID_NETWORK_ACTIVATED:
        Trace(1, "network activate success..");
        OS_ReleaseSemaphore(semMqttStart);
        break;
```

whit this error:

> Assert received!! Reading detail...
> ASSERT DETAIL : 
> I2C resource not active: -1583345645

}

---

pulkin, 2020-02-16:
Try doing it [with python](https://github.com/pulkin/micropython/tree/master/ports/gprs_a9). [Example](https://github.com/pulkin/micropython/blob/master/ports/gprs_a9/examples/example_50_mqtt.py). Please report there if you have troubles re-enabling GPS.

allejp, 2020-02-16:
Thank you but i need to stay with C language.

allejp, 2020-02-16:
I think the current bug was inserted to fix this "It should be that GPS_Close () did not close the serial port, it will be fixed in the next version"
https://github.com/Ai-Thinker-Open/GPRS_C_SDK/issues/139

allejp, 2020-02-16:
Actually this is my code that fix the problem:
```
        UART_Close(UART_GPS);
        OS_Sleep(1000);
        UART_Config_t config = {
            .baudRate = UART_BAUD_RATE_9600,
            .dataBits = UART_DATA_BITS_8,
            .stopBits = UART_STOP_BITS_1,
            .parity = UART_PARITY_NONE,
            .rxCallback = NULL,
            .useEvent = true};
        if (!UART_Init(UART_GPS, config))
            Trace(1, "GPS USART INIT ERROR");
        else
        {
                GPS_Init();
                GPS_Open(NULL);
                // GPS open routine
        }
```

pulkin, 2020-02-16:
Yep, you should consider using the latest 2129 version.

AEmelyanov757, 2020-03-06:
> 
> 
> Yep, you should consider using the latest 2129 version.

When will version 2129 appear ?

zkemble, 2020-03-07:
@AEmelyanov757 There's no official release of 2129, you'll need to download the repo as it is now, but there's a bunch of errors with submodules so to make things easier I've made an unofficial release here https://github.com/zkemble/GPRS_C_SDK/releases/tag/v2.129

pulkin, 2020-03-07:
@zkemble, in case you are interested maintaining GPRS_C_SDK I started tinkering a bit with the binary blobs (elf/lod files). So far I fixed the [this bug](https://github.com/Ai-Thinker-Open/GPRS_C_SDK/issues/262), [here](https://github.com/pulkin/micropython/tree/master/ports/gprs_a9/libcsdk-patches) are rsync patches. I think we can do a lot more given [most of the sources are actually available](https://github.com/cherryding1/RDA8955_W17.44_IDH?).

Btw, I am pretty sure that updating submodules does not change anything for those using csdk functions: the lod file containing them is never built and uploaded directly to the board. I think this applies to GPS stuff as well: those libs are present for header files only.

zkemble, 2020-03-10:
@pulkin A nice thought, but my schedule is already kinda full :P Good work on getting that bug fixed though!

I'm not sure about all the submodules, but cJSON and GPS libraries are indeed compiled and included in the final lod file as I added some Trace()s to them and the messages appeared in the trace window.