Ai-Thinker-Open/GPRS_C_SDK #405: Multiple MQTT_Subscribe call callback with last argument used

allejp, 2020-02-16:
## 1. SDK version（SDK 版本）

{

C_SDK_2108

}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{

Hi, I'm trying to subscribe to two MQTT topics using the same callback function but passing different arguments.

**Expected behaviour:**

> OnMqttSubscribed function called twice with different argument

**Actual behaviour:**

> OnMqttSubscribed function called twice but with the same argument, the last one.

This is my code:
```
        MQTT_SubscribeArg_t *arg1 = (MQTT_SubscribeArg_t *)OS_Malloc(sizeof(MQTT_SubscribeArg_t));
        arg1->client = pEvent->client;
        arg1->topic = POWERMANAGER_TOPIC;

        err = MQTT_Subscribe(pEvent->client, POWERMANAGER_TOPIC, 0, OnMqttSubscribed, (void *)arg1);
        if (err != MQTT_ERROR_NONE)
            Trace(1, "MQTT subscribe error to topic %s, error code:%d", POWERMANAGER_TOPIC, err);
        else
            OS_Free(arg1);

        MQTT_SubscribeArg_t *arg2 = (MQTT_SubscribeArg_t *)OS_Malloc(sizeof(MQTT_SubscribeArg_t));
        arg2->client = pEvent->client;
        arg2->topic = SHADOW_GET_ACCEPTED;

        err = MQTT_Subscribe(pEvent->client, SHADOW_GET_ACCEPTED, 0, OnMqttSubscribed, (void *)arg2);
        if (err != MQTT_ERROR_NONE)
            Trace(1, "MQTT subscribe error to topic %s, error code:%d", SHADOW_GET_ACCEPTED, err);
        else
            OS_Free(arg2);
```

}

---

allejp, 2020-02-16:
My fault, OS_Free(arg1) is in the wrong path.

closed 2020-02-16 by allejp

Soram-jxx, 2020-03-05:
When I used MQTT to subscribe to two topics, I set up the same data receive callback function, and when I sent messages to the two topics using paho clients, the data in the data receive callback function went haywire.

[image]

Soram-jxx, 2020-03-05:
Have you ever had this problem?