Ai-Thinker-Open/GPRS_C_SDK #406: A9G pudding board contains the battery charger module?

allejp, 2020-02-18:
It is not clear to me if the A9G pudding dev board has the battery charger.

In some product details it is specified:

> **There are lithium battery charge management**, microphone, speaker interface, USB communication interface, etc...

So, can I expect to charge the battery if I connect the USB power?

Thanks.

[image]

turbo-lab, 2020-02-26:
Yes! A9G contains ic for charge the battery. It is SY8089 dc-dc converter.

zkemble, 2020-02-26:
No, the SY8089 is not a battery charger. It only converts 5V USB down to
4.2V for the A9G module. Connecting USB and a battery at the same time will
allow an uncontrolled amount of current to flow into the battery, possibly
damaging the battery and converter.
I find that a lot of these Chinese GSM modules (not the dev/carrier board)
say stuff about having battery charging and management, but then never give
any info on how to use it :/

allejp, 2020-02-26:
Ok, so there is not a battery charger, it's correct?
But in [this official page](https://github.com/Ai-Thinker-Open/GPRS_C_SDK/blob/master/README_EN.md) is declared this feature for chipset A9 and A9G:

> Power Management Unit: Lithium battery charge management, integrated DC-DC and LDOs, variable IO voltage

zkemble, 2020-02-26:
There's no charging support, That page also says it has a USB 1.1 interface, but there's nothing about how to use it. Basically, they're listing the features of the main processor (RDA8955L), but some of them might need extra software or hardware support that is missing from the rest of the A9G module and SDK.

turbo-lab, 2020-02-27:
Zak, you are right! Pudding's supply subsystem isn't properly designed. 
A few words about "charging support". I found one interesting cause of power on: POWER_ON_CAUSE_CHARGE. Can it be the way to handle the ending of charge?

allejp, 2020-03-06:
What is the simplest li-ion battery charger solution that we can add to this this board?
I think we need a battery charger with load sharing, can the MCP73831 be useful for this purpose?

allejp, 2020-04-03:
> There's no charging support, That page also says it has a USB 1.1 interface, but there's nothing about how to use it. Basically, they're listing the features of the main processor (RDA8955L), but [...]

Hi Zak, I read your excellent [article](https://blog.zakkemble.net/a-lithium-battery-charger-with-load-sharing/) about lipo charger with load sharing, very informative thank you.
I would like to try to integrate the MCP73831 to the pudding board, I have some doubts on where to connect the VCC output to the [pudding board](http://wiki.ai-thinker.com/_media/gprs/a9g/a9g-dev_v1.2.pdf) to consume as little as possible.
Can you give me some advice?

allejp, 2020-04-04:
Hi Zak, ok I will do so, thank you.

Meanwhile, I found an old board of mine that uses the MCP73831, but it doesn't seem to be in load sharing configuration.

[image]

These are what I misured in VABT e +5V pins:

- USB only
  - VBAT = 4.14V
  - 5V = 4.94V
- Battery only
  - VBAT = 4.04V (real battery voltage I suppose)
  - 5V = 5.00V
- USB with Battery
  - VBAT = 4.14V
  - 5V = 4.54V (seems to increase as the battery charges)

Do you think I can use the power part of this board to poweron the pudding board with load sharing?
In case, wich pins should I connect to pudding board? I think VBAT is a great candidate to use directly in VBAT pins of pudding board.

Thank you.

zkemble, 2020-04-04:
I'm not sure if you really need Q4 on PROG and EN -
PROG: The charger IC will already stop charging when 5V is removed.
EN: If the regulator output is higher than what it is supposed to be then it will stop boosting. The quiescent current is probably low enough not to interfere with battery charging.

But otherwise, what you have there is basically the boost regulator replacing Q1 MOSFET from the load sharer, with a diode (D12) blocking reverse current. You still need to connect the output to VUSB/5V since it will be exceeding 4.2V.

You'll be fine with connecting a charger without load sharing to the battery then the battery to VBAT on the pudding board, the only problem with this is that it will never completely finish charging.

Or did you mean VBAT to pudding VBAT and 5V to pudding VUSB/5V at the same time? Not sure which part you mean by 'power part' either >.<

allejp, 2020-04-05:
Hi Zak, I will try to connect VBAT directly to pudding board VBAT, thank you.

linhphung15999, 2020-09-16:
> It is not clear to me if the A9G pudding dev board has the battery charger.
>
> In some product details it is specified:
>
>
> So, can I expect to charge the battery if I connect the USB power?
>
> Thanks. [...]

I think the A9g Pudding board does support battery charging.
I have a 3.7v lithium battery connected to the board ( (+) -> VBAT, (-) -> GND) and i can charge it through the micro USB port.

Moreover, the board can manage the charging process.
```
typedef enum{
    PM_CHARGER_STATE_DISCONNECTED      = 0,
    PM_CHARGER_STATE_CONNECTED         = 1,
    PM_CHARGER_STATE_CHARGING           = 2,
    PM_CHARGER_STATE_FINISHED          = 3,
    PM_CHARGER_STATE_ERROR_TEMPERATURE = 4,
    PM_CHARGER_STATE_ERROR_VOLTAGE     = 5,
    PM_CHARGER_STATE_ERROR_UNKNOWN     = 9,
    PM_CHARGER_STATE_MAX
}PM_Charger_State_t;

typedef enum{
    PM_BATTERY_STATE_NORMAL     = 0,
    PM_BATTERY_STATE_LOW        = 1,
    PM_BATTERY_STATE_CRITICAL   = 2,
    PM_BATTERY_STATE_SHUTDOWN   = 3,
    PM_BATTERY_STATE_UNKNOWN    = 4,
    PM_BATTERY_STATE_MAX
}PM_Battery_State_t;
```
I found it in the [api_inc_pm.h library](https://github.com/Ai-Thinker-Open/GPRS_C_SDK/blob/master/include/api_inc/api_inc_pm.h) and it works fine!

zkemble, 2020-09-16:
The pudding board does not have any charge management. This schematic shows VBAT connected directly to the 5V -> 4.2V buck regulator output. http://wiki.ai-thinker.com/_media/gprs/a9g/a9g-dev_v1.2.pdf

>Connecting USB and a battery at the same time will allow an uncontrolled amount of current to flow into the battery, possibly damaging the battery and converter.

barokemind, 2020-10-15:
> The pudding board does not have any charge management. This schematic shows VBAT connected directly to the 5V -> 4.2V buck regulator output. http://wiki.ai-thinker.com/_media/gprs/a9g/a9g-dev_v1.2.pdf

hello, I agree with zkemble. I have another problem, when I connect the A9G module with a USB power source, the module starts automatically, but when I use the VBAT pin with a li-ion battery, nothing happens ( i have to turn it on with the board bottom). Do you know why it happens? I checked the schematic and I don't know why.

zkemble, 2020-11-18:
@barokemind Connect PWR_KEY to GND with a piece of wire. What happens is if it detects 5V (USB) at the VUSB_DET pin it will automatically turn on, otherwise if USB is not connected then it will wait for PWR_KEY to go LOW.

ahmadfarisfs, 2020-11-18:
[image]
yes, i can confirm that sy8089 cannot charge the v_bat, only regulates 5v input to v_bat (4.2 - ish) to be used in a9g. SO in conclusion 19g dev board cannot charge battery from usb

Tomblarom, 2021-01-13:
So what are you using for compact / integrated charging solutions?

ens4dz, 2021-08-29:
I found that a9g dev board uses more than 20mA in power off mode !!!
Any hint to save battery from flat discharge ?

turbulator, 2021-08-29:
External p-ch mosfet with power management uC like attiny13a

jmstark, 2022-08-01:
In case it has not been sufficiently clear to anyone:
Do NOT connect a battery and another power source to the board at the same time! It will not charge the battery, but rather destroy the board and maybe even the battery!
(While fiddling around for a while, at one point I forgot to disconnect the battery and a broken board was the result. Luckily I have another A9G pudding board and the battery seems to be working still.)

ricardodanielalbano, 2023-06-16:
I'm using this circuit for my project, a lipo charged with shared load capability. Some parts taken from internet, other from a9g dev board schametics, I have tested it in lab and seems to work fine.

[image]

camp-easy, 2023-09-21:
[image]
Hi everyone, so the vusb-det pin (17) should be left floating in this way? Is my power supply part correct? Thank you in advance...

ricardodanielalbano, 2023-09-22:
Yes, VUSB (17) not connected. About your power supply I have no experience with SY8089, I'm using TPS54308 for downconverting power. I see some mistakes in your design, for example the ufl connector (gps/gprs) is incorrectly connected, the pin 1 is signal, not gnd. Also I guess about ADC0, whats is for? is sensing power supply 3.7v?, also I recommend feed de A9G module with 4v~4.2v, not 3.7v, with 3.7v you're very close to the under limit of Vin and when modul is doing GPRS there are at least 1A of peaks that can drop your 3.7v to lower values reseting the decive. USP_P/USB_N not documented in AI SDK, never used it, I can't imagine how to use it, there isn't documentation. Regards.

camp-easy, 2023-09-22:
Thank you man. I fixed but i would like to feed the A9G processor with 3.8 v so i can use a simple 3.8v lithium battery ... Do you think it will be ok? I have seen the 2 resistors connected in this way with the ADC0 in the A9G dev. official schematic of AiThinker so i do so, even if i didn't understand what they are for. Do you think there are any others potential issues in this part of my schematic? Thank you in advance. Regards.

[image]

ricardodanielalbano, 2023-09-23:
I suggest you to use a lipo charger and a shared load circuit to switch between your external supply and battery when there is no external source present. Take a look at my previus post with a lipo charger and shared load circuit control, I have a new version running on production now with some modifications to improve time of conmutation and fix some worng resistors values, but the idea in the picture is the same.

camp-easy, 2023-09-23:
So the lipo charger shared load circuit is to recharge a lipo battery right? (Sorry but im still an inexpert)

ricardodanielalbano, 2023-09-23:
The two mosfets block is for conmuting between battery and external power supply. The charger is the TP4056 IC.