Ai-Thinker-Open/GPRS_C_SDK #408: download links are not working

DDANPEAL, 2020-02-26:
## 1. SDK version（SDK 版本）

{

The last one

}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{

windows

}

---

Hello,

I'm trying to download the files for Download CSDTK4.2 and SDK but links doesn't seem to work. Only baidu but it needs account for that which I'm avoiding sense I'm not familiar with the site.
Links in https://ai-thinker-open.github.io/GPRS_C_SDK_DOC/en/c-sdk/installation.html
in https://github.com/Ai-Thinker-Open/GPRS_C_SDK/releases
Can you please update the links ? it's really a bad sign to see that getting started guid is not working correctly.

Best regards.

zkemble, 2020-02-26:
Hi, SDK development ended at the end of 2018, so don't expect things to be fixed by the original author (there are still loads of bugs which can't be fixed since most of the SDK is closed source) :(

The MEGA cloud link for the CSDTK download still works - https://mega.nz/#!uKAlRA7Z!UPHn3ZoLY8X33iMIKcBcUYFwrhmNYe-x_dyOdrm_d6I

Also consider using the latest v2.129 SDK, not v2.112. See https://github.com/zkemble/GPRS_C_SDK/releases/tag/v2.129

DDANPEAL, 2020-02-26:
Ok, so no need to start a new project with this module ?

zkemble, 2020-02-26:
That's up to you, really. If it does everything you need it to do right now then no reason not to use it, you'll just have to keep an eye out for SDK bugs (most of which have already been reported here) and figure out a way to work around them.

DDANPEAL, 2020-02-26:
I just want it to send SMS at each movement detection. and go to sleep with the lowest power consumption possible.

zkemble, 2020-02-26:
Yea, there shouldn't be any problems with making that.

DDANPEAL, 2020-02-26:
Thank you Zak, that was helpful

closed 2020-02-26 by DDANPEAL

DDANPEAL, 2020-02-26:
The CSDTK  is not working. it is blocked by my chrome is reporting DLG_FLAGS_SEC_CERT_CN_INVALID
https://mega.nz/#!uKAlRA7Z!UPHn3ZoLY8X33iMIKcBcUYFwrhmNYe-x_dyOdrm_d6I

Is it too much to ask for another link ? maybe a reupload !

zkemble, 2020-02-26:
Seems fine for me (there's probably an option in advanced to continue anyway). Anyhow, I've added CSDTK42_Update2_18_7_27.7z to https://github.com/zkemble/GPRS_C_SDK/releases/tag/v2.129

DDANPEAL, 2020-02-26:
You're the man. thank you very much