Ai-Thinker-Open/GPRS_C_SDK #409: Detected BCPU in the GDB loop，Detected XCPU in the GDB loop 谢谢

onionscode, 2020-03-04:
1.SDK版本 V2.112
2.代码修改了gps_tracker例子，调用gprs提交gps数据。
出现异常。

[image]
[image]
[image]

onionscode, 2020-03-04:
[image]

nghiakuke, 2020-04-19:
I hope to have answer for your ask

nghiakuke, 2020-04-26:
Don't have any answer

marcosasilvalepe, 2020-12-07:
I fixed this by flashing the Factory AT command binary from
https://docs.ai-thinker.com/en/gprs/firmware

with this tool
https://docs.ai-thinker.com/_media/gprs/ai-thinker_a9_a9g_at_v02.02.20190915r.rar