Ai-Thinker-Open/GPRS_C_SDK #410: Protecting code/firmware.

prattom, 2020-03-04:
## 1. SDK version（SDK 版本）

{

V2.112

}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{

How can we protect our code/ firmware in a9? I can't find anything related to this in avilable documentation.

}

---

pulkin, 2020-03-07:
What exactly do you mean?

prattom, 2020-03-07:
I meant to say that after flashing my a9 module is their any way using which any person can get access to or read hex/binary file which was flashed. Is there any to protect firmware/code? Like we have encryption of flash and secure boot on esp32.

prattom, 2020-03-07:
We want to use a9 in commercial device thus we need security for protecting firmware/code/hex/binary flashed in a9.

pulkin, 2020-03-07:
I am pretty sure there is no security there.

zerosigu, 2020-03-19:
Ai-thinker official has encryption tools, please consult relevant people for help.

ji4a, 2024-09-05:
@prattom Did you already found a encryption solution? If yes, how?