Ai-Thinker-Open/GPRS_C_SDK #411: Received UART data but cannot show in trace

benqll, 2020-03-12:
I need to receive the UART data and show up on the trace but i only got the length of the data. Here is what it shows up on the coolwatcher.

[image]

And here is my code which is a simple UART receive testing code. i have tested to use the UART_Read to get the data but with the same problem.

#include <api_os.h>
#include <api_hal_uart.h>
#include <api_debug.h>
#include <string.h>
#include <stdio.h>

#include "api_os.h"
#include "api_event.h"

#define RECEIVE_BUFFER_MAX_LENGTH 200
#define TIMEOUTMS 1000

#define MAIN_TASK_STACK_SIZE    (1024 * 2)
#define MAIN_TASK_PRIORITY      0 
#define MAIN_TASK_NAME         "UART Test Task"

static HANDLE uartTaskHandle = NULL;

uint8_t data[RECEIVE_BUFFER_MAX_LENGTH];
uint32_t timeout = TIMEOUTMS;

static void EventDispatch(API_Event_t* pEvent)
{
    switch(pEvent->id)
    {
        case API_EVENT_ID_UART_RECEIVED:
            Trace(1,"uart received API_EVENT_ID_UART_RECEIVED:%d",pEvent->param1);
            if(pEvent->param1 == UART2)
            {
		//int length;
		//length = UART_Read(UART2,data,length, timeout);
		memcpy(data, pEvent->pParam1, pEvent->param2+1);
                Trace(1,"uart received data,length:%d,data:%s",pEvent->param2,data);
            }
            break;
        default:
            break;
    }
}

static void uart_MainTask()
{
    API_Event_t* event=NULL;
    UART_Config_t config = {
        .baudRate = UART_BAUD_RATE_2400,
        .dataBits = UART_DATA_BITS_8,
        .stopBits = UART_STOP_BITS_1,
        .parity   = UART_PARITY_NONE,
        .rxCallback = NULL,
        .useEvent = true,
    };
    UART_Init(UART1,config);
    UART_Init(UART2,config);
    while(1)
    {
	UART_Write(UART1,"testing",strlen("testing"));

           if(OS_WaitEvent(uartTaskHandle, (void**)&event, OS_TIME_OUT_WAIT_FOREVER))
            {
                EventDispatch(event);
                OS_Free(event->pParam1);
                OS_Free(event->pParam2);
                OS_Free(event);
            }
 
    }
}

void uart_Main(void)
{
    uartTaskHandle = OS_CreateTask(uart_MainTask ,
        NULL, NULL, MAIN_TASK_STACK_SIZE, MAIN_TASK_PRIORITY, 0, 0, MAIN_TASK_NAME);
    OS_SetUserMainHandle(&uartTaskHandle);
}

zkemble, 2020-03-12:
From a quick glance it looks like it should work, however there is a bug with the UART buffers where it seems that both UART1 and 2 use the same buffer. Since you're sending data from UART1 to UART2 it could be messing things up. Try using a USB-serial converter connected to UART2 instead.

Also, `memcpy(data, pEvent->pParam1, pEvent->param2+1);` looks like you're copying an extra byte, hoping for a null to terminate the string, but that extra byte could be anything. You need to use something like `data[pEvent->param2] = '\0';` after the memcpy instead.

benqll, 2020-03-13:
Thank you for replying. 
And now i can get some message from another device but the data will not show up until i power the sender device. Those data mixed up to one string and shown after powered off the sender.
Why would that happen? and How can i fix it?

Johhnzero, 2020-03-16:
when you use USE_EVENT,don't use rxCallback ,choose the true uart number,just only one, UART_Write(UART1,data,pEvent->param2) wiil send that you get.

Johhnzero, 2020-03-16:
[image]

Johhnzero, 2020-03-16:
[image]

zkemble, 2020-03-16:
@Johhnzero It might be usful to post your code for this, since @benqll hasn't used any callback stuff in any of the code they've provided so far.