Ai-Thinker-Open/GPRS_C_SDK #412: Links expired

emeric0101, 2020-03-20:
Hello

All links in https://ai-thinker-open.github.io/GPRS_C_SDK_DOC/en/c-sdk/installation.html 
for downloading CSDK42 are expired, can you help me please ?

zkemble, 2020-03-20:
Hey emeric, I've uploaded CSDK42 here along with the latest SDK stuff since the last official release https://github.com/zkemble/GPRS_C_SDK/releases/tag/v2.129

emeric0101, 2020-03-20:
@zkemble you saved my day !

Thank you

closed 2020-03-20 by emeric0101

gumby0q, 2020-08-21:
there is linux version:
`https://github.com/pulkin/csdtk42-linux`

Benedito821, 2021-05-28:
> Hey emeric, I've uploaded CSDK42 here along with the latest SDK stuff since the last official release https://github.com/zkemble/GPRS_C_SDK/releases/tag/v2.129

@zkemble  is there the windows version of CSDK42?