Ai-Thinker-Open/GPRS_C_SDK #414: [no issue] Some questions about GPS usage

allejp, 2020-03-24:
Hi, this is not a real issue (or at least not about api but documentation).
[GPS documentation](https://ai-thinker-open.github.io/GPRS_C_SDK_DOC/en/c-sdk/function-api/gps.html) is a bit sparse, I have some doubts about the use of GPS API.

1. `GPS_SetFixMode` - What it is supposed to change between these modes?
In the GPS demo is used `GPS_FIX_MODE_LOW_SPEED`, there is a reason?
```
GPS_FIX_MODE_NORMAL = 0,
GPS_FIX_MODE_LOW_SPEED  = 1,
GPS_FIX_MODE_HIGH_SPEED = 2,
GPS_FIX_MODE_ELEVATION  = 3,
```
2. Does `GPS_SetOutputInterval` affects power consumption?
3. Does enabling SBAS (or QZSS) increase power consumption?
4. What is the difference between these power levels?
```
GPS_LP_MODE_NORMAL         = 0,
GPS_LP_MODE_LP             = 8,
GPS_LP_MODE_SUPPER_LP      = 9,
```
5. What are the default values regarding these settings if not set?
```
GPS_SetLpMode
GPS_SetSearchMode
GPS_SetSBASEnable
GPS_SetQzssEnable
GPS_SetFixMode
```

Thanks

CRONOS-IOS, 2020-04-24:
Tienes el código para realizar la publicación en la web  de los datos gps que te saca el modulo A9G?

ahammers, 2020-12-28:
I would also be interested in this information. 

I did a gps tracking firmware which should minimize the delay on the map but the GPS quality is poor and sometimes there are no updates for 5, 10 oder even 30 seconds even if I am using a GPS output interval of 2000ms (GPS data does not change). I am setting GPS_LP_MODE_SUPPER_LP but I do not know if quality and/or update interval is affected by this setting.

So a few questions:
- What are the effects of setting GPS_LP_MODE_SUPPER_LP?
- What are the effects of setting GPS_FIX_MODE_LOW|HIGH_SPEED?