Ai-Thinker-Open/GPRS_C_SDK #416: Burn firmware fails -> bignum too big to convert into long with CSDTK42 Coolwatcher V2.01.03

bartremans, 2020-03-31:
## 1. SDK version（SDK 版本）

{

V2.112

}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{

Not possible to burn firmware:
Verifying fails at first block with error: `[]': bignum too big to convert into `long' (RangeError)
Coolwatcher is version V2.01.03 from CSDTK42
[image]

Also links to official resources (http://test.ai-thinker.com/csdtk, baidu, mega.. ) no longer working?

}

---

roboduck, 2020-04-18:
Hi, I was facing the same error. 
What helped for me was chosing another Flash programmer, not the one from CSDTK42, but there is another one GPRS_C_SDK/platform/ramrun/flsh_spi32m/host_8955_flsh_spi32m_ramrun.lod

You may give a try.

LoneGunnerMe, 2020-08-12:
My attempts to use the .lod under the SDK path failed, but F:\CSDTK42\cooltools\chipgen\modem2G\toolpool\plugins\fastpf\flash_programmers path under host_8955_flsh_spi32m_ The ramrun.lod file is available.