Ai-Thinker-Open/GPRS_C_SDK #417: Bug while making a call

win32cpp, 2020-04-02:
Bug while making a call.
Causes the module to restart.

##############################################
# FATAL assert 
#    --> FINT received before FrameEnd with WinNb 3
##############################################
[16:21:05.240]   SXR 01 : Stack owner not found 0x81c0fd8c\n
[16:21:05.240]   SXR 01 : BitMapReady 0x0
[16:21:05.240]   SXR 01 : Job Monitoring A job (0, 0x15)
[16:21:05.240]   SXR 01 : wait event 14.\n
[16:21:05.240]   SXR 01 : Stack: 760/1000 Top 0x82275f38 <- 0x82276320 Data 0x0\n
[16:21:05.240]   SXR 01 : Job Monitoring B job (1, 0x16)
[16:21:05.241]   SXR 01 : is stopped.\n
[16:21:05.241]   SXR 01 : Stack: 4294967288/1000 Top 0x82265ec0 <- 0x822662a8 Data 0x0\n
[16:21:05.241]   SXR 01 : Job Near job (2, 0x3)
[16:21:05.241]   SXR 01 : wait 1 frames.\n
[16:21:05.243]   SXR 01 : Stack: 408/1000 Top 0x822662c0 <- 0x822666a8 Data 0x0\n
[16:21:05.243]   SXR 01 : Job Cch control job (3, 0x12)
[16:21:05.243]   SXR 01 : is stopped.\n
[16:21:05.243]   SXR 01 : Stack: 460/1000 Top 0x82276338 <- 0x82276720 Data 0x0\n
[16:21:05.250]   SXR 01 : Job Tch job (6, 0xa)
[16:21:05.250]   SXR 01 : wait 1 frames.\n
[16:21:05.250]   SXR 01 : Stack: 716/1000 Top 0x822f3d38 <- 0x822f4120 Data 0x820c2168\n
[16:21:05.250]   SXR 01 : Back trace\n
[16:21:05.250]   SXR 01 : 82005a35y\n
[16:21:05.250]   SXR 01 : 8200595cy\n
[16:21:05.250]   SXR 01 : 88071c0dy\n
[16:21:05.250]   SXR 01 : 82009220y\n
[16:21:05.250]   SXR 01 : 82005a35y\n
[16:21:05.250]   SXR 01 : 82005a15y\n
[16:21:05.250]   SXR 01 : 880aaf75y\n
[16:21:05.250]   SXR 01 : 82008df7y\n
[16:21:05.250]   SXR 01 : 880aaf75y\n
[16:21:05.250]   SXR 01 : 81c07d0dy\n
[16:21:05.250]   SXR 01 : 82005df8y\n
[16:21:05.250]   SXR 01 : 880aafd7y\n
[16:21:05.251]   SXR 01 : 82008df7y\n
[16:21:05.251]   SXR 01 : 81c0541dy\n
[16:21:05.253]   SXR 01 : 81c04eb5y\n
[16:21:05.253]   SXR 01 : 88071c0dy\n
[16:21:05.253]   SXR 01 : 8200d355y\n
[16:21:05.253]   SXR 01 : 81c04eb5y\n
[16:21:05.254]   SXR 01 : 81c02697y\n
[16:21:05.254]   SXR 01 : 88071bd5y\n
[16:21:05.359]   SXR 01 : 81c07ca7y\n

pulkin, 2020-04-03:
You have to give more details like your code, SDK version if you are seeking for help.

win32cpp, 2020-04-04:
I am using the latest available lib for GPRS_C_SDK 2131 and CSDTK 4.2;
Normally there is no problem with one or more module calls(_use CALL_Dial(DIAL_NUMBER) function_), but after an accidental time, the call (_CALL_Dial(DIAL_NUMBER)_) will reset the module.

pulkin, 2020-04-04:
I may be able to help with 2129 only.

ens4dz, 2020-09-15:
You should try this workaround:
https://github.com/Ai-Thinker-Open/GPRS_C_SDK/issues/421
 or force the module to use 1800mhz

win32cpp, 2021-04-08:
Thanks - How i can force the module to use 1800mhz?
any api or method can help me...
**In Normal mode with Call reset module will not be reset only with about 10 to 12 hours in idle mode and call execution, the module will be reset.**