Ai-Thinker-Open/GPRS_C_SDK #420: Cannot create PPP connection to pudding A9G board

jukkar, 2020-04-13:
I am not sure if this is a correct forum, but I did not find better location to get some assistance to the issue I am seeing.

I have a A9G pudding board running running V02.02.20190915R firmware. I am trying to connect that A9G board to other MCU via UART. The idea is that A9G is just a GPRS modem that the other MCU is using. This means that I am using AT commands, that are sent to A9G, to enable PPP mode to connect to Internet. The final PPP handshake fails when A9G sends `peer refused to authenticate` to me. Below is a log output of the issue. The output is devided to two parts, first the AT commands which succeed, and then the PPP handshakes which fails.

```
uart:~$ *** Booting Zephyr OS build zephyr-v2.2.0-1289-gdb18b1770faf  ***

[00:00:00.000,000] <dbg> modem_gsm.gsm_init: Generic GSM modem (0x200004e0)
[00:00:00.000,000] <dbg> modem_gsm.gsm_init: iface->read 0x00010f61 iface->write 0x00010f83
[00:00:00.000,000] <dbg> modem_gsm.gsm_rx: starting
[00:00:00.000,000] <dbg> modem_gsm.gsm_configure: Starting modem 0x200004e0 configuration
[00:00:00.000,000] <dbg> modem_cmd_handler: SENT DATA
41 54                                            |AT               
[00:00:00.000,000] <dbg> net_l2_ppp.net_ppp_init: (0x20002870): Initializing PPP L2 0x2001c89c for iface 0x2001c860
[00:00:00.005,462] <inf> sample_gsm_ppp: main: booted board 'reel_board' APN 'internet' UART 'UART_1' device 0x2001c57c
[00:00:00.403,259] <dbg> modem_cmd_handler: RECV
49 6e 69 74 2e 2e 2e                             |Init...          
[00:00:01.913,696] <dbg> modem_cmd_handler: RECV
2b 43 49 45 56 3a 20 22  43 68 61 72 67 69 6e 67 |+CIEV: " Charging
22 2c 31 30 30 25                                |",100%           
[00:00:02.000,061] <dbg> modem_gsm.gsm_configure: modem not ready -60
[00:00:02.000,061] <dbg> net_l2_ppp.ppp_startup: (0x20002930): PPP 0x2001c89c startup for interface 0x2001c860
[00:00:02.000,091] <dbg> net_l2_ppp.lcp_init: (0x20002930): proto LCP (0xc021) fsm 0x2001c8cc
[00:00:02.000,091] <dbg> net_l2_ppp.ipcp_init: (0x20002930): proto IPCP (0x8021) fsm 0x2001c9b8
[00:00:02.000,122] <dbg> modem_gsm.gsm_configure: Starting modem 0x200004e0 configuration
[00:00:02.000,122] <dbg> modem_cmd_handler: SENT DATA
41 54                                            |AT               
[00:00:02.031,097] <dbg> modem_cmd_handler: RECV
4f 4b                                            |OK               
[00:00:02.031,127] <dbg> modem_cmd_handler.cmd_handler_process: match cmd [OK] (len:2)
[00:00:02.031,127] <dbg> modem_gsm.gsm_cmd_ok: ok
[00:00:02.031,219] <dbg> modem_cmd_handler: SENT DATA
41 54 2b 43 4f 50 53 3d  30 2c 30                |AT+COPS= 0,0     
[00:00:02.069,732] <dbg> modem_cmd_handler: RECV
2b 43 4d 45 20 45 52 52  4f 52 3a 20 33          |+CME ERR OR: 3   
[00:00:04.032,348] <err> modem_gsm: AT+COPS ret:-60
[00:00:04.032,348] <dbg> modem_cmd_handler: SENT DATA
41 54 45 30                                      |ATE0             
[00:00:04.061,401] <dbg> modem_cmd_handler: RECV
4f 4b                                            |OK               
[00:00:04.061,401] <dbg> modem_cmd_handler.cmd_handler_process: match cmd [OK] (len:2)
[00:00:04.061,401] <dbg> modem_gsm.gsm_cmd_ok: ok
[00:00:04.111,572] <dbg> modem_cmd_handler: SENT DATA
41 54 48                                         |ATH              
[00:00:04.139,434] <dbg> modem_cmd_handler: RECV
4f 4b                                            |OK               
[00:00:04.139,434] <dbg> modem_cmd_handler.cmd_handler_process: match cmd [OK] (len:2)
[00:00:04.139,434] <dbg> modem_gsm.gsm_cmd_ok: ok
[00:00:04.189,605] <dbg> modem_cmd_handler: SENT DATA
41 54 2b 43 4d 45 45 3d  31                      |AT+CMEE= 1       
[00:00:04.217,681] <dbg> modem_cmd_handler: RECV
4f 4b                                            |OK               
[00:00:04.217,681] <dbg> modem_cmd_handler.cmd_handler_process: match cmd [OK] (len:2)
[00:00:04.217,712] <dbg> modem_gsm.gsm_cmd_ok: ok
[00:00:04.267,852] <dbg> modem_cmd_handler: SENT DATA
41 54 2b 43 47 4d 49                             |AT+CGMI          
[00:00:04.296,478] <dbg> modem_cmd_handler: RECV
41 69 5f 54 68 69 6e 6b  65 72 5f 43 6f 2e 5f 4c |Ai_Think er_Co._L
74 64 2e                                         |td.              
[00:00:04.296,508] <dbg> modem_cmd_handler.cmd_handler_process: match cmd [] (len:19)
[00:00:04.296,539] <inf> modem_gsm: Manufacturer: Ai_Thinke
[00:00:04.297,027] <dbg> modem_cmd_handler: RECV
4f 4b                                            |OK               
[00:00:04.297,027] <dbg> modem_cmd_handler.cmd_handler_process: match cmd [OK] (len:2)
[00:00:04.297,027] <dbg> modem_gsm.gsm_cmd_ok: ok
[00:00:04.347,229] <dbg> modem_cmd_handler: SENT DATA
41 54 2b 43 47 4d 4d                             |AT+CGMM          
[00:00:04.375,762] <dbg> modem_cmd_handler: RECV
41 39 2f 41 39 47                                |A9/A9G           
[00:00:04.375,762] <dbg> modem_cmd_handler.cmd_handler_process: match cmd [] (len:6)
[... 126 lines cut ...]
[00:00:05.031,341] <dbg> net_l2_ppp.fsm_recv_terminate_req: (0x20002494): [LCP/0x2001c8cc] Terminated by peer (peer refused to authenticate)
[00:00:05.031,341] <dbg> net_l2_ppp.ppp_change_state_debug: (0x20002494): [LCP/0x2001c8cc] state OPENED (9) => STOPPING (5) (fsm_recv_terminate_req():846)
[00:00:05.031,372] <dbg> net_l2_ppp.ppp_fsm_lower_down: (0x20002494): [IPCP/0x2001c9b8] Current state REQUEST_SENT (6)
[00:00:05.031,372] <dbg> net_l2_ppp.ppp_change_state_debug: (0x20002494): [IPCP/0x2001c9b8] state REQUEST_SENT (6) => STARTING (1) (ppp_fsm_lower_down():239)
[00:00:05.031,402] <dbg> net_l2_ppp.ppp_fsm_close: (0x20002494): [IPCP/0x2001c9b8] Current state STARTING (1)
[00:00:05.031,402] <dbg> net_l2_ppp.ppp_change_state_debug: (0x20002494): [IPCP/0x2001c9b8] state STARTING (1) => INITIAL (0) (ppp_fsm_close():211)
[00:00:05.031,433] <dbg> net_l2_ppp.ppp_change_phase_debug: (0x20002494): [0x2001c89c] phase NETWORK (3) => DEAD (0) (ppp_link_down():118)
[00:00:05.031,433] <dbg> net_l2_ppp.validate_phase_transition: (0x20002494): Invalid phase transition: NETWORK (3) => DEAD (0)
[00:00:05.031,433] <dbg> net_l2_ppp.ppp_change_phase_debug: (0x20002494): [0x2001c89c] phase DEAD (0) => ESTABLISH (1) (lcp_down():235)
[00:00:11.038,452] <dbg> net_l2_ppp.net_pkt_hexdump: recv L2
[00:00:11.038,513] <dbg> net_l2_ppp: 0x20019ef8
c0 21 05 04 00 20 70 65  65 72 20 72 65 66 75 73 |.!... pe er refus
65 64 20 74 6f 20 61 75  74 68 65 6e 74 69 63 61 |ed to au thentica
74 65                                            |te               
[00:00:11.038,543] <dbg> net_l2_ppp.ppp_fsm_input: (0x20002494): [LCP/0x2001c8cc] LCP Terminate-Req (5) id 4 payload len 28
[00:00:11.038,543] <dbg> net_l2_ppp.fsm_recv_terminate_req: (0x20002494): [LCP/0x2001c8cc] Current state STOPPING (5)
[00:00:11.038,543] <dbg> net_l2_ppp.fsm_recv_terminate_req: (0x20002494): [LCP/0x2001c8cc] Invalid state STOPPING (5)
[00:00:15.031,524] <dbg> net_l2_ppp.ppp_fsm_timeout: (0x20002930): [LCP/0x2001c8cc] Current state STOPPING (5)
[00:00:15.031,555] <dbg> net_l2_ppp.ppp_change_state_debug: (0x20002930): [LCP/0x2001c8cc] state STOPPING (5) => STOPPED (3) (ppp_fsm_timeout():105)
[00:00:15.031,555] <dbg> net_l2_ppp.ppp_change_phase_debug: (0x20002930): [0x2001c89c] phase ESTABLISH (1) => DEAD (0) (ppp_link_terminated():103)
[00:00:15.031,555] <dbg> net_l2_ppp.ppp_link_terminated: (0x20002930): [0x2001c89c] Link terminated
uart:~$ 
```

Any idea why A9G terminates the connection establishment? It ACKs the connection attempt at time 04.985,504 but then later terminates the connection. There is no authentication on PPP link, so no PAP or similar is used. The APN does not have any password or username. The message `peer refused to authenticate` is confusing as there is no need to authenticate anywhere. The same setup works works fine if I use SIMCOM SIM808 modem instead of A9G, so the cellular connection setup is correctly set.

kebaranas, 2020-08-08:
Same problem. I also cannot use PPP from Mongoose OS.

kebaranas, 2020-08-08:
I was able to get it to work. If your user and password is empty ("", ""). Add a space (" ", " ") to the user and pasword. I use the same module and version but different os. Mine is Mongoose OS.