Ai-Thinker-Open/GPRS_C_SDK #421: Pudding board power issues with 850/900MHz GSM band

zkemble, 2020-04-15:
The A9G has a transmit power of 2W on the 850/900MHz bands and 1W on the 1800/1900MHz bands. When connected to a mobile network on 850/900MHz the current draw of the module can be up to 1.8A in short bursts (you will need a current shunt resistor and oscilloscope to see the bursts, an ammeter is too slow). On the pudding board these high currents create large voltage drops on the VBAT supply due to insufficient decoupling capacitance. There is only a 100uF tantalum capacitor near the module VBAT pin and a 22uF ceramic capacitor at the buck converter.

**VBAT:**
[image]

**VUSB:**
[image]

These 2.3V drops cause all kinds of problems, the main ones being random crashes and restarts and SD card problems. Reading/writing to the SD card might fail and the SD card might completely lock up and require a power cycle to fix it.

I added 2x 470uF electrolytic capacitors (should be low ESR, but the ones I used were not) and a 1uF ceramic capacitor in parallel with the 100uF tantalum capacitor:

[image]

This reduced the voltage drops down to 0.6V, which seems to have fixed most of the issues, but it's still not great.

**VBAT:**
[image]

**VUSB:**
[image]

The buck regulator used on the pudding is a SY8089, rated for 2-3A. The inductor will need to have a saturation current rating of at least 1.95A for a 1.8A load (I think I worked that out right), however the 2.2uH inductor appears to be slightly physically smaller than inductors with such a rating and thus might have a lower saturation rating, which could be another cause for the output voltage dropping.

On the 1800/1900MHz band everything is fine without the extra capacitors, as the burst current is only around 0.8A with a small voltage drop of around 200mV, and then a bit of overshoot.

**VBAT:**
[image]

**VUSB:**
[image]

So, if you're having weird problems with your pudding board you could try:
- Using a mobile provider that operates on the 1800/1900MHz GSM band (in the UK that’s EE and any other network that piggyback on their signal), this one is the easiest and most reliable fix.
- Adding a bunch of capacitors.
- Power it from a USB port that can supply a lot of juice, laptop USB ports tend to be pretty weak.
- Use short and good quality USB cables.
- Power the board directly from a good lithium battery at the VBAT pins. Don't connect USB power if you do this, see https://github.com/Ai-Thinker-Open/GPRS_C_SDK/issues/406#issuecomment-591341069
- Change the inductor on the buck regulator.

Though, there is one good thing that comes from using the 850/900MHz band; the microphone picks up far less GSM interference!

zkemble, 2020-04-22:
So I had a look at work for some inductors and borrowed the closest I could find to what the pudding board has, [this one](https://uk.farnell.com/wurth-elektronik/7447786006/pd-inductor-6050-6-8uh-sat-2-3a/dp/2082714). 6.8uH with 2.5A saturation current.

[image]

VBAT:
[image]

The inductor is an improvement over the previous inductor, 1.11V vs 2.27V, but still not as good as adding the capacitors.

I added back the 2x 470uF and 1x 1uF ceramic capacitors.

VBAT:
[image]

This is about the same as with the original 2.2uF inductor. I'm guessing that the extra capacitance was smoothing out the current peaks so the 2.2uH inductor was able to handle the current and didn't saturate.

I swapped out the 2x 470uF capacitors with 2x low impedance 470uF caps ([these](https://uk.farnell.com/rubycon/6-3zlh470mefc6-3x11/cap-470-f-6-3v-20/dp/8127174)), this reduced the drops a little bit more, now at 0.57V.

[image]

So, the inductor only helps if no extra capacitors are added. However, adding extra capacitors works better than changing the inductor.

ahammers, 2021-01-08:
If I power the Pudding board directly by a 2800mAh 18650 Liion cell connected to VBAT, is there still a power problem? 

Where did you add the capacitors? To which net is the onboard capacitor connected to? 

Thank you and I am sorry if this is a stupid question. I do not know enough about electronics to evaluate this myself. I just want to build a stable gps tracker with low latency for our next race.

sdml0, 2024-01-02:
For stable work with somekind 5v power supply(using built-in 4.2v converter) at very least 1000uF is required on VBAT.
There is even convinient pads available close to usb connector.