Ai-Thinker-Open/GPRS_C_SDK #423: GPS Issue

emeric0101, 2020-04-20:
## 1. SDK version（SDK 版本）

{

GPRS_C_SDK_V2129 and master on git

}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{

With the Demo GPS, the device is stuck on "gprs register complete"
I am working on Windows 10, in an open area, the device is an a9g pudding with the included gps antenna. I use a 10W external usb power supply in France

}

Can you help me to find the issue please
---

zkemble, 2020-04-20:
Try adding more Trace()s around the code to see exactly where it is getting stuck.

emeric0101, 2020-04-27:
It seems to be stuck just after : 
GPS_Open(NULL);
in the loop :     while(gpsInfo->rmc.latitude.value == 0)

The event API_EVENT_ID_GPS_UART_RECEIVED never occurs

Thank

emeric0101, 2020-04-27:
What I received : 
[image]

zkemble, 2020-04-27:
Looks like you're getting something there. Try other baud rates, like 115200, 19200, 57600 etc.

emeric0101, 2020-04-29:
I have try to flash (with the Ai-thinker tools) the GPS firmware and it failed. I guess it is an issue from the board. I contact the seller

Thank for your help

CheetahGod, 2020-05-19:
I have the same problem. IT says waiting for gprs registers to complete

CheetahGod, 2020-05-19:
[trace.txt](https://github.com/Ai-Thinker-Open/GPRS_C_SDK/files/4653410/trace.txt)