Ai-Thinker-Open/GPRS_C_SDK #424: APN with two chars HALT the device and the workaround with space is not allowed by my network

brunohorta82, 2020-04-21:
i need use the apn "wm" and if i use "wm " the device can register on the network, you can fix this to accept only two chars?

zkemble, 2020-04-21:
I think @pulkin managed to fix this in their micropython implementation, see issue https://github.com/pulkin/micropython/issues/28

You'll need to patch SW_V2129_csdk.lod with this https://github.com/pulkin/micropython/commit/8ce363a91468fefde82beec34aa473a59bd93d41

or this might be more useful https://github.com/pulkin/micropython/tree/master/ports/gprs_a9/libcsdk-patches

brunohorta82, 2020-05-06:
ok but I have all code in C :(,

pulkin, 2020-05-07:
Up-to-date patches are here:

https://github.com/pulkin/micropython/blob/8f1b0a32f724f4a42fffbfcc3aba05086ea1da27/ports/gprs_a9/libcsdk-patches/patch-lod.py#L6-L11

I hope they are self-explanatory.