Ai-Thinker-Open/GPRS_C_SDK #425: Broken download links for CSDTK

XaviDCR92, 2020-04-28:
## 1. SDK version（SDK 版本）

{

Should not matter, but I'm using the latest commit on this repo so far (8d4426f32df699594f42abeddb99f43c82d36f45)

}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{
Kubuntu 18.04.4 LTS
}

---

As the title suggests, all three links from https://ai-thinker-open.github.io/GPRS_C_SDK_DOC/en/c-sdk/installation_linux.html are down ([baidu](https://pan.baidu.com/s/1FzEvkt9z41QdvUdLxZJvug), [MEGA](https://mega.nz/#!LHhxWTrJ!rR7RzMxOOq4zwfGhpVLfe7DmdwEfhIiTLArb8OFqJM0) and [direct link](http://test.ai-thinker.com/csdk/CSDTK42_Linux.tar.gz)). Is there any other link I can get CSDTK from?

Thank you.

zkemble, 2020-04-28:
Check here! -> #412

XaviDCR92, 2020-04-28:
Thanks for the blazing fast answer, @zkemble !

XaviDCR92, 2020-04-29:
@zkemble the link you attached only seems to include Windows binaries. Is there any version for Linux?

zkemble, 2020-04-29:
Sorry, I don't have the Linux CSDTK, and I just noticed you wanted the Linux version in your first post, not Windows! Hopefully someone who does have the Linux stuff comes by and sees this...

pulkin, 2020-05-04:
https://github.com/pulkin/csdtk42-linux

XaviDCR92, 2020-05-04:
> https://github.com/pulkin/csdtk42-linux

Thank you very much, @pulkin !

closed 2020-05-04 by XaviDCR92