Ai-Thinker-Open/GPRS_C_SDK #427: CSDTK4.2 download links are all dead...

marcosasilvalepe, 2020-05-18:
I just started interacting with this really cool module but I can't compile and try the different demo files because I can't download the tools from the installation url (https://ai-thinker-open.github.io/GPRS_C_SDK_DOC/en/c-sdk/installation.html).

The links are all dead...

Where else can I download the files ??

## 1. SDK version（SDK 版本）

{

}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{

}

---

zkemble, 2020-05-18:
Check here! -> #412

marcosasilvalepe, 2020-05-18:
Thank you very much!!! :)

closed 2020-05-18 by marcosasilvalepe

DomolinBolivia, 2021-03-04:
https://drive.google.com/drive/folders/17lOulAhnx3yaaq3liTwQsLRtwp2CL7JJ?usp=sharing