Ai-Thinker-Open/GPRS_C_SDK #428: Does my schema contains any potentional issues?

glararan, 2020-05-23:
I would like to ask before I do manufacturing of my board about A9G.

I have following schema:
[image]

(No need to check battery management section, power supply)

What it should do:
- Power on automatically (even on battery)
- Make a call even on battery (18650)
- Custom firmware upload via DEBUG headers
- A9G custom firmware to read [SPI RFID card module](https://www.ebay.com/itm/I2C-SPI-Interface-IC-RC522-RFID-Read-Write-Card-Module-13-56MHz-IC-Keychain/142899334766?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649)
- Read M1, M2 (pull-down)
- Output LED_RFID, BUZZER

Questions:
- Will all selected IOs work for described features? I cant find GPIO logic if there is some problem / special pin
 - M1, M2 Inputs
 - LED_RFID, BUZZER Outputs
 - SPI
- Can A9G work with 3.3V?
- Is A9G blocked during call? I mean if I can still read incoming RFID card during call

zkemble, 2020-05-23:
Have you tried prototyping any code with the pudding dev board? You may find things to be a little bit quirky and weird.
You have some SPI stuff, be careful of #332.
You have insufficient decoupling capacitance if you use an 850/900MHz mobile provider, see #421.
You have the USB data lines connected, but USB comms is not supported (won’t be a problem, just an FYI).
A9G needs 3.5 - 4.2V. 3.3V is too low and the AP7363 regulator recommended max current is 1.5A, but the module might need up to 2A.
The battery should be directly connected to the module without diodes (D2 + D3), otherwise the supply voltage will be too low.
Calls do not block, but be careful of these problems #331 and #248.
U3 and Q3 are not doing anything.

glararan, 2020-05-23:
@zkemble 
Nope, not yet.

Thanks for info about SPI. I might not need to talk via SPI at booting moment. 
Yep Im gonna use 900MHz. Thanks
USB - read about it. I know its useless. Programming is via HW serial

I am using A9 chip. Main power should be 5V via VUSB pin. If Main power fails then VBAT goes up. AP7363 is there only to get correct voltage for RFID / Buzzer. I meant in my question if GPIO works at 3.3V. My bad

Ok I'll remove D3. D2 Is needed to not mix 5V node with for entire circuit. 

Call - Im not using mic, just using module to ring phone as alert.

U3/Q3 suppose to disconnect battery if too low :)

zkemble, 2020-05-23:
The VUSB pin is not for powering the module, it's only used to powering the USB interface and to detect when USB is connected, so it doesn't really need to be connected.

3.3V for A9 GPIO is fine then. Though, the regulator is quite over-spec’d for an RFID module, datasheet says it takes around 150mA max.

You should use something like this instead of diodes on the VBAT supply https://blog.zakkemble.net/a-lithium-battery-charger-with-load-sharing/

It would be best to prototype making phone calls, I think I've read that some people have problems with it (especially when using a 900MHz network, it eats A LOT of current).

Q3 sources are all connected to ground, it does nothing. One source is supposed to go to circuit GND and the other source goes to battery negative terminal. The DW01 GND goes to battery negative, not circuit GND, same for the 0.1u capacitor.

glararan, 2020-05-23:
Ah.. ok.. I thought VUSB can act like +5V pin power supply pin. Well then. Should I remove mapping to pin and go just with VBAT?

Yeah I know.. however I replaced it with 500mA regulator (schema below).

I dont see problem with eating a lot of current :) PSU is capable of 20W which is enough. 18650 should handle it aswell for ringing max 3 times on battery per hour.

Btw thanks for checking DW01. You were right! (Should be right now). 

[image]

zkemble, 2020-05-23:
VUSB can be left connected, it doesn't really make a difference in your case.

I just notied that T1 is a PSU, I thought it was a bridge rectifier, that should be ok then xP

VBAT still goes through a diode to the regulator, have you checked the voltage drop of the diode? Looks like it's at least 0.7V, so when the battery hits ~4V the regulator output will begin to drop.

DW01 circuit looks good now :)

glararan, 2020-05-24:
Yeah youre right about that voltage drop :/ MOSFET would be nice there.. 

[image]
datasheet - https://datasheet.lcsc.com/szlcsc/1810301520_ON-Semiconductor-FDN336P_C10882.pdf

well then this could be working schema. Thanks

zkemble, 2020-05-24:
That won't work, the MOSFET will always be on. Vgs must be greater than around -0.2V.
When powered by PSU: Vgs = 0V - (5V - D4drop) = -4.3V = ON.
When PSU power is removed: Vgs = 0V - (Vbat - Q1drop) = -3.5V = ON.

There's also no input capacitance to the regulator.

glararan, 2020-05-24:
[image]

Now it should be fine. If i didnt do any more mistake :/

zkemble, 2020-05-24:
Looks ok now, however when power is removed from the PSU the voltage might only drop slowly, where the MOSFET won't fully turn on for a brief moment and battery current will flow through Q1 internal diode creating a drop, but might not be a problem.

BL8568 recommends 1uF on the input and output, not 0.1uF.

Just noticed that 5V goes to the buck regulator, then the 4.2V bypasses the charger IC and goes straight to the battery. This is super bad, as uncontrolled current will flow into the battery and possibly damage it.

glararan, 2020-05-24:
Wrong image.. sorry :/ I put there 1uF. 

Yeah there was SS34 before at 
[image]
. Wheres the best place to put diode?

zkemble, 2020-05-24:
Get rid of all the diodes and MOSFETs, follow this https://blog.zakkemble.net/a-lithium-battery-charger-with-load-sharing/ and put the PSU on VIN and everything else on the "To load" output. Your TP4056 can be used instead of the MCP73831.

glararan, 2020-05-24:
@zkemble 

Is this correct design?
[image]

zkemble, 2020-05-24:
D1 will bypass the 4.2V regulator. U4 VIN is disconnected.
Move the 4.2V regulator to the output of the load sharing, then the output of the regulator should be VBAT. Problem with this is that when on battery the regulator will cause a small voltage drop, but it shouldn't be too bad, less than a diode.

For best performance, D1 should be a MOSFET and there should be some active control circuitry added to decide when the MOSFETs are on and off. Then the 4.2V regulator can be moved to between Q2 gate and D1 (or Q since D1 would be changed to a MOSFET). But that's for another time...

glararan, 2020-05-24:
D1: oh yes :/ I see now. 
U4: shit.. forgot to connect it

If there will be huge 3V3 voltage drop then SPI module will just reboot. No big deal.

Updated schema (Hope so I didnt forget anything):
[image]

zkemble, 2020-05-24:
Yea that looks good now. I think the worst that might happen is the A9 resets when removing PSU power, as power switches from PSU to battery.
R6 should be 100k, not 100.

glararan, 2020-05-24:
@zkemble 
Thanks man a lot. Yeah R6 fixed. I think about same during power switch. I think its not important as long as A9 boots up. There will be no memory sensitive data. However it will be dissapointing to register into network again..

zkemble, 2020-05-25:
Just a thought, maybe a P-MOSFET across the 4.2V regulator with source connected to VREG, drain to +4V and gate to +5V would work to bypass it when powered by battery.

glararan, 2020-05-25:
I am thinking more of pulling gate to ground on Q2. Then MOSFET should work all time. MOSFETs diode will not allow charging battery from 5V node if im correct?

zkemble, 2020-05-25:
No, the internal diode is parallel to the MOSFET, and MOSFETs can conduct in any direction when on.

glararan, 2020-05-25:
Well then it not great :D Only way I see is to put proper diode (I dont have much experience in MOSFETs).

zkemble, 2020-05-25:
A proper diode where?
Nows your chance to get some experience with MOSFETs :P

glararan, 2020-05-25:
There instead MOSFET.
[image]

Diode allows VBAT current flow to VREG but 5V/VREG node will not affect VBAT. However diode will drop voltage at the end :/
I am thinking if its problem at all or not. Because SY8089 can take lower input voltage and still boost it to 4.2V, 4.2V node goes properly to 3V3 voltage regulator. Everything should be kinda fine. Except I am missing knowledge about what could go wrong :D

zkemble, 2020-05-25:
Again, the diode will create too much of a voltage drop. Thats why a MOSFET is used instead, to get rid of that drop. SY8089 is not a boost regulator, it can only drop/buck/step-down the voltage.

glararan, 2020-05-25:
Yeah. I think I'll stay with current concept as it should work. I will let you know if PSU disconnecting results into A9 reboot.

glararan, 2020-09-18:
@zkemble Thank you for helping with circuit. Removing PSU while battery is on is not restarting A9 and keeps going. Circuit works fine!

closed 2020-09-18 by glararan

bduunity, 2022-03-24:
How to interface RFID RC522 with A9G module