Ai-Thinker-Open/GPRS_C_SDK #433: MQTT SSL doesn't work with latest csdk-lib commit

faithware, 2020-06-07:
With the last csdk lib commit 8095f01aa67a88378cad9482082e2e1383907674
aka version 2131 .
MQTT SSL doesn't work, connect messages are sent as if they are mqtt messages.

This product can be a huge change maker and only if you guys release at least the socket and mqtt librairies this will take of to the next level.
Look at the mqtt implementaion of the ESP32 for instance, it is absolutely fantastic. Why can't we do something like that?
I have issues with the mqtt client on the module being very instable with some mqtt brokers. 
Please do something
Bests

zkemble, 2020-06-07:
This might help https://github.com/Ai-Thinker-Open/GPRS_C_SDK/issues/323#issuecomment-630148181

However, it seems that A9G development stopped at the end of 2018. If you want fixes, official support, bug-free libraries and stability then the A9/G is not something you'll want to use.
v2131 from Johhnzero looks like a messed-up fork of this repo (anyone can just make up a version number too).

faithware, 2020-06-09:
Thanks @zkemble 

I have seen that the rda8955 SDK was released but I can't understand why ai thinker just gave up on this product. I this the A9 is reliable and versatile but the software is instable. 
Not to mention the debugging capabilities and the flash programming tools, almost all the competition in this space are not that far away. 
I would love a real explanation why are they doing this.