Ai-Thinker-Open/GPRS_C_SDK #438: will my new schema work?

pagman, 2020-06-30:
Hello, i have being playing for a while now with the a9g pudding board and now i want to make my own pcb design for a simple gps tracker (not led's, io etc), my plan is with the debug headers to upload my own firmware other than that it will be a simple pcb. 

i would appriciate if you can give me your thoughts on my design and if you think it is gonna work.
i am sorry in advance because it is my first design schema so it might not look profetional at all

[image]

MakhmudSultonov, 2020-07-12:
[image]

pagman, 2020-07-12:
thank you i have completely overlook the power key ! 
i am thinking to use a tantalum capacitor fot the 100uf  and a ceramic for the 100nf  for the decoupling you think they will be fine? Thank you

zkemble, 2020-07-14:
R1 and R2 are not needed. The A9 can already read the battery voltage using PM_Voltage(). See this #421 about your decoupling, 100u might be ok since you're directly connected to a battery and not dealing with an SD card.

pagman, 2020-07-14:
> R1 and R2 are not needed. The A9 can already read the battery voltage using PM_Voltage(). See this #421 about your decoupling, 100u might be ok since you're directly connected to a battery and not dealing with an SD card.

Thank you for the help! I thought I need the voltage devider even for using the PM_Voltage(), because I saw this on the schematic.
[image]

zkemble, 2020-07-14:
I'm not sure why they have it there, maybe it just makes it easier to test the external ADC stuff. This project of mine #396 can still read the supply voltage without it.

pagman, 2020-07-14:
> I'm not sure why they have it there, maybe it just makes it easier to test the external ADC stuff. This project of mine #396 can still read the supply voltage without it.

thank you i really appriciate you help! i think my questions about the schematic are covered i am marking it as closed

closed 2020-07-14 by pagman